Freescale/meta-freescale #1215: linux-fslc-imx fails to compile for imx6qdlsabresd with newest Yocto Kirkstone

dv1, 2022-09-17:
When I try to build an image with current Yocto Kirkstone, I see that `linux-fslc-imx` fails to compile. The error from `log.do_compile` is:

```
arm-oel-linux-gnueabi-ld.bfd: arch/arm/mach-imx/busfreq-imx.o: in function `busfreq_probe':
/usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1384: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1379: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1361: undefined reference to `init_freq_optee'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1361: undefined reference to `init_freq_optee'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1362: undefined reference to `update_freq_optee'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1362: undefined reference to `update_freq_optee'
arm-oel-linux-gnueabi-ld.bfd: arch/arm/mach-imx/busfreq-imx.o: in function `enter_lpm_imx6_up':
/usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:234: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: arch/arm/mach-imx/busfreq-imx.o: in function `bus_freq_pm_notify':
/usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1062: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/arch/arm/mach-imx/busfreq-imx.c:1070: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk.o: in function `imx_clk_disable_uart':
/usr/src/kernel/drivers/clk/imx/clk.c:199: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk-composite-8m.o: in function `imx8m_clk_hw_composite_flags':
/usr/src/kernel/drivers/clk/imx/clk-composite-8m.c:227: undefined reference to `imx_src_is_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk-gate2.o:/usr/src/kernel/drivers/clk/imx/clk-gate2.c:58: more undefined references to `imx_src_is_m4_enabled' follow
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk-imx8mm.o: in function `imx8mm_clocks_probe':
/usr/src/kernel/drivers/clk/imx/clk-imx8mm.c:337: undefined reference to `check_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk-imx8mn.o: in function `imx8mn_clocks_probe':
/usr/src/kernel/drivers/clk/imx/clk-imx8mn.c:332: undefined reference to `check_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk-imx8mp.o: in function `imx8mp_clocks_probe':
/usr/src/kernel/drivers/clk/imx/clk-imx8mp.c:724: undefined reference to `check_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/clk/imx/clk-imx8mq.o: in function `imx8mq_clocks_probe':
/usr/src/kernel/drivers/clk/imx/clk-imx8mq.c:322: undefined reference to `check_m4_enabled'
arm-oel-linux-gnueabi-ld.bfd: drivers/gpu/drm/bridge/nwl-dsi.o: in function `nwl_dsi_get_lcm':
/usr/src/kernel/drivers/gpu/drm/bridge/nwl-dsi.c:961: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: drivers/gpu/drm/bridge/sec-dsim.o: in function `sec_mipi_dsim_calc_pmsk':
/usr/src/kernel/drivers/gpu/drm/bridge/sec-dsim.c:1192: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: drivers/firmware/imx/imx-scu-soc.o: in function `imx_scu_soc_init':
/usr/src/kernel/drivers/firmware/imx/imx-scu-soc.c:117: undefined reference to `TKT340553_SW_WORKAROUND'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/firmware/imx/imx-scu-soc.c:117: undefined reference to `TKT340553_SW_WORKAROUND'
arm-oel-linux-gnueabi-ld.bfd: drivers/staging/media/imx/hdmirx/cdns-hdmirx-hw.o: in function `hdmirx_config':
/usr/src/kernel/drivers/staging/media/imx/hdmirx/cdns-hdmirx-hw.c:880: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.o: in function `pma_pll_config':
/usr/src/kernel/drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.c:1059: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.o: in function `clk_ratio_detect':
/usr/src/kernel/drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.c:1430: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.c:1432: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.c:1434: undefined reference to `__aeabi_uldivmod'
arm-oel-linux-gnueabi-ld.bfd: drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.o:/usr/src/kernel/drivers/staging/media/imx/hdmirx/cdns-hdmirx-phy.c:1436: more undefined references to `__aeabi_uldivmod' follow
```

This seems related to these patches:

https://source.codeaurora.org/external/imx/linux-imx/commit/?h=lf-5.15.y&id=d1e806347e888e13d6fdddb28275dbce7fd9f780
https://source.codeaurora.org/external/imx/linux-imx/commit/?id=b199c5a4709cd5681c7b740829c77780730e6898

I will next try to build the image with `linux-imx` instead to test if that one builds.

`meta-freescale` is at commit `3068a16ff6fde4d025f234332b953456e93cc8f6`, "Merge pull request #1210 from thochstein/gtk+3".

otavio, 2022-09-18:
Which defconfig did you use?\

dv1, 2022-09-18:
The default one that is in meta-freescale. I did not use anything custom.

otavio, 2022-09-18:
And which machine?

dv1, 2022-09-18:
`imx6qdlsabresd`

angolini, 2022-09-20:
I can reproduce it on my side. And I'm testing changes in the defconfig. When I have something I update thisissue

otavio, 2022-09-21:
@dv1 please give it a go. @thochstein please check this as the driver seems to lack a Kconfig depends which would select the driver to provide the required symbols.

dv1, 2022-10-03:
Confirmed, it works now. Built an image with the updated kernel successfully.

otavio, 2022-10-03:
I am then closing this.

closed 2022-10-03 by otavio