Freescale/meta-freescale #1216: Vulkan support for iMX8mm

munoz0raul, 2022-09-19:
Hi, I'm trying the meta-flutter on my distro with Wayland/Weston.

It worked well when using `backend-egl`, but I heard that should work even better if I use `backend-vulkan`.

Base on that information, I added to my Image:

vulkan-tools
vulkan-loader
vulkan-headers

So, when I tried to run vulkaninfo, I got:

```
root@imx8mm-lpddr4-evk:/var/rootdirs/home/fio# vulkaninfo 
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
ERROR at /usr/src/debug/vulkan-tools/1.3.204.1-r0/git/vulkaninfo/vulkaninfo.h:649:vkCreateInstance failed with ERROR_INCOMPATIBLE_DRIVER
```

After some search, I found this post: https://community.toradex.com/t/vulkan-api-returns-vk-error-incompatible-driver-9/11858/2

Based on it, I created the same file just fixing the path. Not sure about api_version as well.

```
cat /etc/vulkan/icd.d/vivante.json
{
   "file_format_version": "1.0.0",
   "ICD": {
      "library_path": "/usr/lib/libvulkan_VSI.so.1",
      "api_version": "1.1.176"
   }
}
```

Well, this made the vulkaninfo start to work but the flutter is falling with a Segmentation fault.

```
vulkaninfo      
'DISPLAY' environment variable not set... skipping surface info
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2: Emulation cannot generate unique IDs for struct VkPhysicalDeviceIDProperties - setting IDs to zero instead
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
WARNING: [Loader Message] Code 0 : vkGetPhysicalDeviceProperties2KHR: Emulation found unrecognized structure type in pProperties->pNext - this struct will be ignored
[... 511 lines cut ...]
	multiviewTessellationShader = false

VkPhysicalDeviceProtectedMemoryFeatures:
----------------------------------------
	protectedMemory = false

VkPhysicalDeviceSamplerYcbcrConversionFeatures:
-----------------------------------------------
	samplerYcbcrConversion = false

VkPhysicalDeviceShaderDrawParametersFeatures:
---------------------------------------------
	shaderDrawParameters = false

VkPhysicalDeviceVariablePointersFeatures:
-----------------------------------------
	variablePointersStorageBuffer = false
	variablePointers              = false
```

Should Vulkan work fine with imx-gpu-viv? 

Any suggestion on how to configure it?

I don't know if the flutter segmentation fault helps on anything, if so, let me know and I will share it here.

thochstein, 2022-09-19:
Hi @munoz0raul, sorry, but it seems there is no Vulkan support for 8M Mini. I will try to update the meta-data so this is better handled.

thochstein, 2022-09-19:
See #1218.

closed 2022-09-21 by munoz0raul