Freescale/meta-freescale #1227: linux-fslc-imx fails to compile with patches with newest Yocto Kirkstone

einarjon, 2022-09-23:
I have another layer that builds linux-fslc-imx-5.15.bb with some patches (latest linux-fslc-imx-5.15.67+gitAUTOINC+845d53f13a-r0).

If I add **any**  \*.cfg or \*.patch file to the SRC_URI, the build fails
Without patches the build is fine...
Using MACHINE=imx8qm-mek and kas:3.1

Minimal example to reproduce:
cd /path/to/meta-freescale # or the extra layer
echo 'CONFIG_FOOBAR=y' > recipes-kernel/linux/linux-fslc-imx/foobar.cfg
echo 'SRC_URI:append = " file://foobar.cfg "' > recipes-kernel/linux/linux-fslc-imx_%.bbappend

build fails with
```
| DEBUG: Searching for foobar.cfg in paths:
... snip ...
| DEBUG: Executing shell function do_kernel_configme
| NOTE: do_kernel_metadata: for summary/debug, set KCONF_AUDIT_LEVEL > 0
| ${sccs} = '/build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg'
| $patches = ''
| handling: /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg
| [ERROR]: processing of file /tmp/tmp.24coO353fI failed
| /build/tmp/hosttools/dirname: missing operand
| Try '/build/tmp/hosttools/dirname --help' for more information.
| ERROR: Could not generate configuration queue for imx8qm-mek.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx_5.15.bb:do_kernel_configme) failed with exit code '1'
```

Contents of file /tmp/tmp.24coO353fI (the one that failed)
```
#
# spp v0.8
# processed: Fri 23 Sep 2022 12:52:01 PM UTC
#
# This is a preprocessor output file, do not edit
#
reloc_dir /work/meta-freescale/recipes-kernel/linux
#
prefix /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/mx8-nxp-bsp/defconfig
kconf non-hardware /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/mx8-nxp-bsp/defconfig
prefix /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg
kconf non-hardware /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg
# run time: 0 seconds
# processed files:
# _cfg /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/mx8-nxp-bsp/defconfig
# _cfg /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg
```

angolini, 2022-09-23:
As far as I know, the  `linux-fslc-imx` is not prepared to work with config fragments. 

How does it fail when you apply a patch?

einarjon, 2022-09-24:
Exactly the same with patches. Both patches and fragments worked in hardknott, but now I'm moving to kirkstone and this is an issue.

I actually traced it to poky, so maybe I'm posting in the wrong repo.
https://git.yoctoproject.org/poky/tree/meta/classes/kernel-yocto.bbclass?h=kirkstone#n292
This line is failing if either the variable `${sccs}` (config fragments) or `${patches}` is not an empty string.

I will need to investigate further on Monday.

einarjon, 2022-09-26:
Definitely posted in the wrong repo. Sorry for that.
Kas issue: https://groups.google.com/g/kas-devel/c/cz_HzB5LO4Q

The problem is a bad combination of poky's [scc](https://git.yoctoproject.org/yocto-kernel-tools/tree/tools/scc) not handling relative paths (like /build/../work/meta-freescale)
and [kas](https://github.com/siemens/kas) 3.1 not using real paths. like it did in earlier versions. 

Issue solved by downgrading kas, which lists the file as /work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg instead of /build/../work/meta-freescale/recipes-kernel/linux/linux-fslc-imx/foobar.cfg

closed 2022-09-26 by einarjon