Freescale/meta-freescale #1256: core-image-minimal doesn't prompt login on i.MX6UL EVK

otavioborges, 2022-10-09:
I recently build core-image-minimal for the i.MX6UL Evk on dunfell, to boot through SD card.

On the serial console U-boot and kernel loads ok. But I never get to the login, linux hangs on "random: dd: uninitialized urandom read (512 bytes read)" and doesn't accept any inputs.

Unfortunately I don't have a LCD screen to check if console login is there. I have attached local.conf and kernel dump.

Any help on this would very appreciate :)
[kernel_dump.txt](https://github.com/Freescale/meta-freescale/files/9742128/kernel_dump.txt)
[local.txt](https://github.com/Freescale/meta-freescale/files/9742131/local.txt)

otavio, 2022-10-09:
@otavioborges could you check if it works in kirkstone/master?

otavioborges, 2022-10-09:
Hey @otavio

Ends up it was my own mistake. I am working on a MCIMX6UL-BB, but the SoM is a ULL.
So changing the machine to imx6ullevk did the trick.

I did one more compilation still using imx6ulevk while on kirkstone, in that case U-boot would crash while loading the environment. But I imagine this would be expected trying to use the worng arch :D

otavioborges, 2022-10-09:
I did had another minor issue. After booting the board reached login normally. I got distracted and left without logging and soon after I got an kernel oops
I have attached the console dump
[kernel_oops.log](https://github.com/Freescale/meta-freescale/files/9742456/kernel_oops.log)

otavio, 2022-10-09:
Please open a specific issue for it; I am closing this as the issue has been solved.

closed 2022-10-09 by otavio