Freescale/meta-freescale #1262: Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants)

otavio, 2022-10-12:


angolini, 2022-10-13:
linux-fslc-imx is merged with mainline stable tags up to v5.15.60

https://github.com/Freescale/linux-fslc/tree/5.15-2.1.x-imx

otavio, 2022-10-13:
Are you going to send the PR for the kernel?

angolini, 2022-10-13:
I'm going to send the PR to bump the linux-fslc-imx to 5.15-2.1.x-imx up to v5.15.60 today or tomorrow. I don't have more work done on this at least until end of next week

zandrey, 2022-11-11:
Hello @otavio 

For the Mainline Kernel and U-Boot, I've prepared 2 branches in my fork:
- https://github.com/zandrey/linux-fslc/tree/6.0.x+fslc
- https://github.com/zandrey/u-boot-fslc/tree/2022.10+fslc

Can you please pull those branches in, so I can then do respective recipes upgrade?

Thanks a lot!

-- andrey

otavio, 2022-11-11:
@zandrey, I just pushed both forks.

zandrey, 2022-11-11:
> @zandrey, I just pushed both forks.

Thanks, I'll test my changes locally and would send out PR to cover Kernel and U-Boot, presumably tomorrow.

PR #1294 (2022-11-14): Upgrade Mainline BSP Kernel + U-Boot
This PR bumps the version of both _Kernel_ (both stable and LTS recives) and _U-Boot_ to the latest version that is present in respective forks.

Kernel branch was updated up to and including following versions for recipes from stable korg:
- `linux-fslc_6.0.bb`: _v6.0.8_
- `linux-fslc-lts_5.15.bb`: _v5.15.78_

_U-Boot_ has been upgraded to `v2022.10` tag from _DENX_ repository and all local fork patches were migrated from previous version to a new one.

Due to Kernel OE Sanity reports with respect to missing or invalid config options, there are 2 additiona [... cut]
closed 2022-11-14 by otavio