Freescale/meta-freescale #1267: imx8mm-lpddr4-evk: does not boot out-of-the box using core-image-weston and IMX_DEFAULT_BSP = "nxp"

angolini, 2022-10-14:
How to reproduce:

branch = master
DISTRO ?= 'fslc-wayland'
add `IMX_DEFAULT_BSP = "nxp"` to local.conf
build core-image-weston

**revA**
```
U-Boot SPL 2022.07+fslc+ga21d7668eb (Jul 22 2022 - 15:45:20 +0000)
write error to device: 9311d8 register: x!write error to device: 9311d8 register: x!write error to device: 9311d8 register: x!write error to device: 9311d8 register: x!write error to device: 9311d8 regid
Normal Boot
WDT:   Started watchdog@30280000 with servicing (60s timeout)
Trying to boot from MMC1
NOTICE:  BL31: v2.6(release):lf-5.15.32-2.0.0-0-gc6a19b1a3-dirty
NOTICE:  BL31: Built : 06:37:22, Jun  7 2022
```
**revB**
```
U-Boot SPL 2022.07+fslc+ga21d7668eb (Jul 22 2022 - 15:45:20 +0000)           
SEC0:  RNG instantiated                                                      
Normal Boot                                                                  
WDT:   Started watchdog@30280000 with servicing (60s timeout)                
Trying to boot from MMC1                                                     
NOTICE:  BL31: v2.6(release):lf-5.15.32-2.0.0-0-gc6a19b1a3-dirty             
NOTICE:  BL31: Built : 06:37:22, Jun  7 2022                                 
```

angolini, 2022-10-14:
@otavio , can you please take a look?

dv1, 2022-10-22:
I also have this issue. However, I do have a rev B board.

dv1, 2022-10-24:
More updates: I tried to revert all the way back to u-boot-fslc 2021.07 - the problem persisted. I couldn't revert back to 2021.04 due to too many conflicts. It is of course possible that some other configs are the cause of the problem, and not u-boot itself, but perhaps this still helps.

angolini, 2022-10-26:
`U-Boot SPL 2022.04+fio+g727d16723a (Oct 10 2022 - 06:30:09 +0000)
power_bd71837_init
DDRINFO: start DRAM init
DDRINFO: DRAM rate 3000MTS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
SPL: Unsupported Boot Device!
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
`

otavio, 2022-10-26:
Maybe check if imx-atf needs upgrade as well?

otavio, 2022-11-14:
@angolini please check if master fixes this. @zandrey has sent the upgrade for U-Boot and this could help on this.

angolini, 2022-11-16:
```
U-Boot SPL 2022.10+fslc+g2c2dc1affb (Nov 11 2022 - 09:01:01 +0000)
No pmic                                 
SEC0:  RNG instantiated                 
Normal Boot                             
WDT:   Started watchdog@30280000 with servicing (60s timeout)
Trying to boot from MMC1                
NOTICE:  BL31: v2.6(release):android-12.1.0_1.0.0-0-gc688807dd-dirty
NOTICE:  BL31: Built : 09:52:38, Sep  5 2022
```

And then, the watchdog resets the board.
Where does this "android" tag come from?

dv1, 2022-12-18:
With current meta-freescale git master (at commit ff311ff0cd65616055b2f3c030b65a567a76dcd5 ), u-boot-fslc still does not boot. Addings `IMX_DEFAULT_BOOTLOADER:mx8mm-nxp-bsp = "u-boot-imx"` to `local.conf` helps as a workaround - the image then boots.

otavio, 2022-12-19:
@dv1 please paste full boot log. @angolini tested and it worked for her.

dv1, 2022-12-27:
@otavio Just did a full rebuild of git master (meta-freescale HEAD is at 7b150784237b1b7edc9e9db123da98219fe61132 ), still does not work. This is all I see over UART:

```
U-Boot SPL 2022.10+fslc+gcbd7eb17ac (Dec 13 2022 - 00:42:37 +0000)
No pmic
SEC0:  RNG instantiated
Normal Boot
WDT:   Started watchdog@30280000 with servicing (60s timeout)
Trying to boot from MMC1
NOTICE:  BL31: v2.6(release):android-12.1.0_1.0.0-0-gc688807dd-dirty
NOTICE:  BL31: Built : 09:52:38, Sep  5 2022
```

angolini, 2023-01-17:
@otavio @dv1 I tested it at that time. Is there anything else I should test?

otavio, 2023-01-17:
@dv1 could you retest? We did 2023.01 as well.

MrCry0, 2023-01-17:
@otavio

On my RevB board u-boot-fslc 2023.01 seems to boot well:
```
U-Boot SPL 2023.01+fslc+g3b524eeb24 (Jan 16 2023 - 14:43:15 +0000)
No pmic
SEC0:  RNG instantiated
WDT:   Started watchdog@30280000 with servicing every 1000ms (60s timeout)
Trying to boot from USB SDP
SDP: initialize...
SDP: handle requests...
Downloading file of size 883952 to 0x40400000... done
Jumping to header at 0x40400000
Header Tag is not an IMX image
Found header at 0x40421c00
NOTICE:  BL31: v2.6(release):lf-5.15.71-2.2.0-0-g3c1583ba0-dirty
NOTICE:  BL31: Built : 11:00:38, Nov 21 2022

U-Boot 2023.01+fslc+g3b524eeb24 (Jan 16 2023 - 14:43:15 +0000)

CPU:   Freescale i.MX8MMQ rev1.0 at 1200 MHz
Reset cause: POR
Model: FSL i.MX8MM EVK board
DRAM:  2 GiB
Core:  169 devices, 23 uclasses, devicetree: separate
WDT:   Started watchdog@30280000 with servicing every 1000ms (60s timeout)
MMC:   FSL_SDHC: 1, FSL_SDHC: 2
Loading Environment from MMC... MMC: no card present
*** Warning - No block device, using default environment

In:    serial@30890000
Out:   serial@30890000
Err:   serial@30890000
SEC0:  RNG instantiated
Net:   eth0: ethernet@30be0000
Hit any key to stop autoboot:  0 
MMC: no card present
switch to partitions #0, OK
mmc2(part 0) is current device
...
```

MrCry0, 2023-01-17:
@otavio u-boot-fslc 2023.01 seems to boot well on may RevB:
```
U-Boot SPL 2023.01+fslc+g3b524eeb24 (Jan 16 2023 - 14:43:15 +0000)
No pmic
SEC0:  RNG instantiated
WDT:   Started watchdog@30280000 with servicing every 1000ms (60s timeout)
Trying to boot from MMC2
NOTICE:  BL31: v2.6(release):lf-5.15.71-2.2.0-0-g3c1583ba0-dirty
NOTICE:  BL31: Built : 11:00:38, Nov 21 2022

U-Boot 2023.01+fslc+g3b524eeb24 (Jan 16 2023 - 14:43:15 +0000)

CPU:   Freescale i.MX8MMQ rev1.0 at 1200 MHz
Reset cause: POR
Model: FSL i.MX8MM EVK board
DRAM:  2 GiB
Core:  169 devices, 23 uclasses, devicetree: separate
WDT:   Started watchdog@30280000 with servicing every 1000ms (60s timeout)
MMC:   FSL_SDHC: 1, FSL_SDHC: 2
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial@30890000
Out:   serial@30890000
Err:   serial@30890000
SEC0:  RNG instantiated
Net:   eth0: ethernet@30be0000
Hit any key to stop autoboot:  0 
MMC: no card present
switch to partitions #0, OK
mmc2(part 0) is current device
Scanning mmc 2:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
1:      i.MX8M Mini EVK
Retrieving file: /extlinux/../Image
append: root=/dev/mmcblk1p2 rootwait rw console=ttymxc1,115200
Retrieving file: /extlinux/../imx8mm-evk.dtb
Moving Image from 0x40480000 to 0x40600000, end=42a30000
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
Working FDT set to 43000000
   Loading Device Tree to 000000004fff4000, end 000000004fffffd3 ... OK
Working FDT set to 4fff4000

Starting kernel ...
```

otavio, 2023-01-17:
I will close this and if @dv1 reproduces it again, it can be reopened.

closed 2023-01-17 by otavio