Freescale/meta-freescale #1273: [kirkstone] weston failed to build test due to missing header dependency

bmx666, 2022-10-20:
I created `weston_%.imx.bbappend`

```
PACKAGECONFIG:remove = "headless image-jpeg shell-desktop shell-fullscreen shell-ivi"
PACKAGECONFIG:append = " rdp "
``` 

It cause build issue

```
| FAILED: tests/test-internal-screenshot.p/internal-screenshot-test.c.o                                                                                                                                     
| arm-fslc-linux-gnueabi-gcc -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/build/yocto/nxp-bsp-platform/build-XXX-XXX-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0/recipe-sysroot -Itests/test-internal-screenshot.p -Itests -I../git/tests -I. -I../git -Iinclude -I../git/include -Ilibweston -I../git/libweston -Iprotocol -I/home/build/yocto/nxp-bsp-platform/build-XXX-XXX-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0/recipe-sysroot/usr/include/pixman-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -std=gnu99 -Wmissing-prototypes -Wno-unused-parameter -Wno-shift-negative-value -Wno-missing-field-initializers -Wno-pedantic -Wundef -fvisibility=hidden -DENABLE_EGL -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/build/yocto/nxp-bsp-platform/build-XXX-XXX-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0=/usr/src/debug/weston/10.0.0.imx-r0 -fdebug-prefix-map=/home/build/yocto/nxp-bsp-platform/build-XXX-XXX-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0=/usr/src/debug/weston/10.0.0.imx-r0 -fdebug-prefix-map=/home/build/yocto/nxp-bsp-platform/build-XXX-XXX-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0/recipe-sysroot= -fdebug-prefix-map=/home/build/yocto/nxp-bsp-platform/build-XXX-XXX-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0/recipe-sysroot-native= -DUNIT_TEST '-DTHIS_TEST_NAME="test-internal-screenshot"' -MD -MQ tests/test-internal-screenshot.p/internal-screenshot-test.c.o -MF tests/test-internal-screenshot.p/internal-screenshot-test.c.o.d -o tests/test-internal-screenshot.p/internal-screenshot-test.c.o -c ../git/tests/internal-screenshot-test.c                                      
| In file included from ../git/tests/internal-screenshot-test.c:31:                                                                                                                                         
| ../git/tests/weston-test-client-helper.h:40:10: fatal error: weston-screenshooter-client-protocol.h: No such file or directory                                                                            
|    40 | #include "weston-screenshooter-client-protocol.h"                                                                                                                                                 
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                 
| compilation terminated.
```

but at **build/build.ninja**
```
246:build protocol/weston-screenshooter-client-protocol.h: CUSTOM_COMMAND ../git/protocol/weston-screenshooter.xml | /home/build/yocto/nxp-bsp-platform/build-forza6x-forza-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0/recipe-sysroot-native/usr/bin/wayland-scanner
247: COMMAND = /home/build/yocto/nxp-bsp-platform/build-forza6x-forza-xwayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/weston/10.0.0.imx-r0/recipe-sysroot-native/usr/bin/wayland-scanner client-header ../git/protocol/weston-screenshooter.xml protocol/weston-screenshooter-client-protocol.h
```

I can manually generate `protocol/weston-screenshooter-client-protocol.h` without any problem, but `test-internal-screenshot` does not depend on it.

bmx666, 2022-10-20:
I found commit in official weston repo - https://github.com/wayland-project/weston/commit/2ac6b6b084a877adde64db7faff2ed22eb3ea97a

otavio, 2022-10-24:
Could you prepare a PR for master fixing this?

PR #1278 (2022-10-24): weston: fix build test-internal-screenshot
tests: Add dependency on screenshooter client protocol

Given that the test-helper code relies on the screenshooter protocol, make sure it's available for us to build, and the dependency ensures we build in order.

Link: https://github.com/wayland-project/weston
Commit: 2ac6b6b084a877adde64db7faff2ed22eb3ea97a

Fixes: #1273
commit 69e013b2 (2022-10-24): weston: fix build test-internal-screenshot

tests: Add dependency on screenshooter client protocol

Given that the test-helper code relies on the screenshooter protocol,
make sure it's available for us to build, and the dependency ensures we
build in order.

Link: https://github.com/wayland-project/weston
Commit: 2ac6b6b084a877adde64db7faff2ed22eb3ea97a

Fixes: #1273
commit d2e7b242 (2022-10-24): weston: fix build test-internal-screenshot

tests: Add dependency on screenshooter client protocol

Given that the test-helper code relies on the screenshooter protocol,
make sure it's available for us to build, and the dependency ensures we
build in order.

Link: https://github.com/wayland-project/weston
Commit: 2ac6b6b084a877adde64db7faff2ed22eb3ea97a

Fixes: #1273
closed 2022-10-24 by otavio