Freescale/meta-freescale #1288: imx8qxp-mek-dom0.dtb - No rule to make target 'arch/arm64/boot/dts/freescale/imx8qxp-mek-dom0.dtb'

petero5, 2022-11-09:
Hi

Have followed the instructions here:
https://github.com/Freescale/fsl-community-bsp-platform

But am getting:
| make[1]: *** No rule to make target 'arch/arm64/boot/dts/freescale/imx8qxp-mek-dom0.dtb'.  Stop.
| make: *** [/workdir/fsl-community-bsp/build/tmp/work-shared/imx8qxp-mek/kernel-source/Makefile:1425: freescale/imx8qxp-mek-dom0.dtb] Error 2

i.e.
repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b kirkstone
repo sync
MACHINE=imx8qxp-mek DISTRO=fslc-wayland source setup-environment build
bitbake core-image-minimal

bitbake -e | grep PREFERRED_PROVIDER_virtual/kernel
gives
PREFERRED_PROVIDER_virtual/kernel="linux-fslc-imx"

changing local.conf to
PREFERRED_PROVIDER_virtual/kernel="linux-imx"
(then deleting tmp and rerunning bitbake) produces the same error

See also
https://lists.yoctoproject.org/g/meta-freescale/message/24885

Thank you
-- Peter

otavio, 2022-11-09:
@thochstein I think the machine needs an update due to the Linux kernel upgrade. Could you take a look?

thochstein, 2022-11-09:
Will do.

commit 44c85e6a (2022-11-10): imx8qxp-mek.conf: Drop invalid dtbs

Remove dtbs dropped by linux-imx 5.15.52:
- imx8qxp-mek-dom0.dtb
- imx8qxp-mek-inmate.dtb
- imx8qxp-mek-root.dtb

Fixes: #1288
commit 10466585 (2022-11-10): imx8qxp-mek.conf: Drop invalid dtbs

Remove dtbs dropped by linux-imx 5.15.52:
- imx8qxp-mek-dom0.dtb
- imx8qxp-mek-inmate.dtb
- imx8qxp-mek-root.dtb

Fixes: #1288

(cherry picked from commit 44c85e6a9c4673c22f9a174f4799ee9a35d56d29)
closed 2022-11-10 by otavio