Freescale/meta-freescale #1318: u-boot-fslc 2022.10 failed with exit code '137' (using too much memory)

Livius90, 2022-11-29:
New `u-boot-fslc_2022.10.bb` is failed always with [exit code '137'](https://www.containiq.com/post/exit-code-137) which means the available memory has been exhausted, so 16 GB DDR4 RAM memory was not enough for this single recipe build on Ubuntu 20.04 desktop Linux.

Similar problem was happened in mender forum: https://hub.mender.io/t/uboot-auto-configure-exit-code-137/3624

Please fix it, i think it is it is not normally that 16 GB RAM is not enough for a single recipe build.

basmevissen, 2022-11-29:
You know you can reduce the memory consumption by:
1) Lowering the number of parallel makes (using PARALLEL_MAKE in local.conf or maybe in a .bbappend for the package)
2) Limit the number of bitbake threads (using BB_NUMBER_THREADS in local.conf)
3) Create a swap file on an SSD?

I use the latter with Android AOSP builds as they really need more than 16GB RAM by default.

Livius90, 2022-11-30:
```
PARALLEL_MAKE = "-j1"
```

Can not help in local.conf, it is failed with exit code '137'!

kevswims, 2022-11-30:
I recently ran into an issue with a particular defconfig option for uboot-imx that was causing it to consume massive amounts of memory during compilation until it crashed. I did not end up figuring out exactly what the particular config option was that caused it but you could try using a different config to see if that fixes the issue.  Uboot is not that heavy of a build memory wise so several threads with 16GB of RAM should not be an issue.

Livius90, 2022-11-30:
I can confirm that the high memory usage occurse always at the Kconfig parsing and processing, something is really buggy in u-boot 2022.10 make configuration because my custom defconfig was perfect with 2022.04 version before.

Livius90, 2022-12-01:
I have a custom board which uses a PICO-IMX7 SoM. The default u-boot [pico-pi-imx7d_defconfig](https://github.com/Freescale/u-boot-fslc/blob/2021.04%2Bfslc/configs/pico-pi-imx7d_defconfig) works well, so i try to comparing my custom defconfig with this.

Following configs were missed in my config:
```
CONFIG_HAS_BOARD_SIZE_LIMIT=y
CONFIG_BOARD_SIZE_LIMIT=715776
CONFIG_SPL_MAX_SIZE=0xe000
CONFIG_SPL_HAS_BSS_LINKER_SECTION=y
CONFIG_SPL_BSS_MAX_SIZE=0x100000
CONFIG_SYS_SPL_MALLOC=y
CONFIG_SYS_MAXARGS=32
CONFIG_SYS_PBSIZE=532
```

I placed this config in to my custom defconfig and it solved the memory usage problem. So if one of them missing from that list, it can cause a memory usage issue.

closed 2022-12-01 by Livius90

trini, 2024-06-26:
Just for future reference, when this error code happens, the problem is that the defconfig being used needs to be manually run through "make oldconfig" or similar and the missing options set. This will happen when updating versions of U-Boot, especially when moving from much older versions where we had not finished our migration of `CONFIG` symbols to being in Kconfig only.