Freescale/meta-freescale #1319: Dunfell

ycnakajsph, 2022-11-29:
Already posted on NXP forum but they said dunfell is not officially supported, is it supported here?

Here the link to NXP forum : https://community.nxp.com/t5/i-MX-Processors/Yocto-Dunfell-GSTREAMER-build-fails-imx6qdlsabreauto-fsl/m-p/1537008#M196319 and here's the issue duplicated :

Hi,

There's an issue with gstreamer build with imx6qldsabreauto and fsl-framebuffer _[SAME HAPPENS WITH FSLC]_ distro.

I've been reproducing with upstream bsp. Find the repo complete info here : https://pastebin.com/abwRtKAY

    meta-freescale : ae2001935fdd7cf7ee7e3b8a4826d7b59202be6d

    fsl-community-bsp-base : 5a551f453260bd19895e4d847877874eaa51fde3

    meta-freescale-3rdparty : f37f72a85ba273b27347169b727c889f42b91d6c

    meta-freescale-distro : 5d882cdf079b3bde0bd9869ce3ca3db411acbf3b

    meta-openembedded : 6792ebdd966aa0fb662989529193a0940fbfee00

    poky : 90a6f6a110ab14890e2f6a1616e74ee259fc0f8f

Steps to reproduce :

    git clone https://github.com/Freescale/fsl-community-bsp-platform.git

    cd fsl-community-bsp-platform/

    repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b dunfell

    repo sync

    DISTRO=fsl-framebuffer MACHINE=imx6qdlsabreauto source ./setup-environment build

    bitbake fsl-image-machine-test

Relevant part of the failure of gstreamer1.0-plugins-base_1.16.imx :

    | Package gstreamer-audio-1.0 was not found in the pkg-config search path.
    | Perhaps you should add the directory containing `gstreamer-audio-1.0.pc'
    | to the PKG_CONFIG_PATH environment variable
    | No package 'gstreamer-audio-1.0' found
    | Package gstreamer-video-1.0 was not found in the pkg-config search path.
    | Perhaps you should add the directory containing `gstreamer-video-1.0.pc'
    | to the PKG_CONFIG_PATH environment variable
    | No package 'gstreamer-video-1.0' found

How can I solve this issue?

Best regards,

Yann

angolini, 2022-11-29:
Dunfell is not supported anymore since Kirkstone is the new LTS.

otavio, 2022-11-29:
@ycnakajsph please try to reproduce this using `fslc-framebuffer` distro.

closed 2022-12-07 by otavio