Freescale/meta-freescale #1324: imx6qdlsabresd: MMC/SD card is not recognized when building current git master

dv1, 2022-12-09:
I built an image today with current git master (meta-freescale is at 6f6b3a65548969de0097153f13d1dfd690b9be54 ).

The UART terminal shows:

```
U-Boot SPL 2022.10+fslc+g2c2dc1affb (Nov 11 2022 - 09:01:01 +0000)
Trying to boot from MMC1

U-Boot 2022.10+fslc+g2c2dc1affb (Nov 11 2022 - 09:01:01 +0000)

CPU:   Freescale i.MX6DL rev1.1 at 792MHz
CPU:   Commercial temperature grade (0C to 95C) at 25C
Reset cause: POR
Model: Freescale i.MX6 DualLite SABRE Smart Device Board
I2C:   ready
DRAM:  1 GiB
Core:  74 devices, 19 uclasses, devicetree: separate
PMIC:  PFUZE100 ID=0x10
MMC:   FSL_SDHC: 1, FSL_SDHC: 2, FSL_SDHC: 3
Loading Environment from MMC... MMC: no card present
*** Warning - No block device, using default environment

No panel detected: default to Hannstar-XGA
Display: Hannstar-XGA (1024x768)
In:    serial
Out:   serial
Err:   serial
SEC0:  RNG instantiated
Net:   eth0: ethernet@2188000
Normal Boot
Hit any key to stop autoboot:  0 
MMC: no card present
MMC Device 0 not found
no mmc device at slot 0
```

No matter what I do, I can't boot from the SD card. But reverting commit 53f06469f542cab60f7fb0acee32f519726b1106 fixed this.

MaxKrummenacher, 2022-12-10:
There is [this](https://github.com/u-boot/u-boot/commit/f827f84d3f5607d0b33e927f6127a888e7bd664f) commit from @fabioestevam for Wandboard and [this](https://lore.kernel.org/all/20221208181552.36921-1-detlev.casanova@collabora.com/) is on the mailinglist for Sabrelite. Both addressing a similar sounding issue.

otavio, 2022-12-10:
@dv1 I made a PR (https://github.com/Freescale/u-boot-fslc/pull/50). Could you confirm it fixes your boot issue?

fabioestevam, 2022-12-10:
On Sat, Dec 10, 2022 at 5:45 PM Otavio Salvador ***@***.***>
wrote:

> @dv1 <https://github.com/dv1> I made a PR (Freescale/u-boot-fslc#50
> <https://github.com/Freescale/u-boot-fslc/pull/50>). Could you confirm it
> fixes your boot issue?
>

No, this change is not correct.

It has two issues:

1. It does not build
2. It points to the incorrect mmc device

I submitted the correct fix to the U-Boot list.

Message ID: ***@***.***>
>

otavio, 2022-12-10:
@fabioestevam I applied your patch. Thanks.

dv1, 2022-12-18:
I can confirm this works for me. Thanks!

PR #1325 (2022-12-14): u-boot-fslc: Bump revision to cbd7eb17ac1
Apply following changes:

cbd7eb17ac1 imx8mn-venice-u-boot: Fix broken boot
f463aeddbbf imx8mn-ddr4-evk-u-boot: Fix broken boot
d1920093874 imx6qdl-sabresd: Pass mmc alias
047ac33f481 wandboard: Pass mmc aliases

Fixes: #1324, #1267
commit 37efa651 (2022-12-14): u-boot-fslc: Bump revision to cbd7eb17ac1

Apply following changes:

cbd7eb17ac1 imx8mn-venice-u-boot: Fix broken boot
f463aeddbbf imx8mn-ddr4-evk-u-boot: Fix broken boot
d1920093874 imx6qdl-sabresd: Pass mmc alias
047ac33f481 wandboard: Pass mmc aliases

Fixes: #1324, #1267
commit a99bfd03 (2022-12-14): u-boot-fslc: Bump revision to cbd7eb17ac1

Apply following changes:

cbd7eb17ac1 imx8mn-venice-u-boot: Fix broken boot
f463aeddbbf imx8mn-ddr4-evk-u-boot: Fix broken boot
d1920093874 imx6qdl-sabresd: Pass mmc alias
047ac33f481 wandboard: Pass mmc aliases

Fixes: #1324, #1267

(cherry picked from commit 37efa65108d367fd28cb0aa6ae789670e7358ae7)