Freescale/meta-freescale #1337: imx6: VDOA broken?

dv1, 2022-12-18:
I was trying to use the VDOA to detile frames that are produced by the VPU decoder. I use gstreamer-imx for this purpose, for example with this simple line:

    gst-launch-1.0 filesrc location=test.mp4 ! qtdemux ! h264parse ! imxvpudec_h264 ! imxg2dvideosink

`imxvpudec_h264` uses libimxvpuapi, which in turn uses the IPU (and in turn the VDOA) to detile the frames. This worked flawlessly in the past.

But now, I see in kirkstone and master this in dmesg:

```
[  126.720041] ERROR: v4l2 capture: slave not found!
[  127.372534] 8<--- cut here ---
[  127.375606] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  127.383764] pgd = a9ee5276
[  127.386589] [00000000] *pgd=00000000
[  127.390224] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[  127.395547] Modules linked in: 8021q caam_jr caamkeyblob_desc caamhash_desc caamalg_desc crypto_engine rng_core authenc libdes mxc_v4l2_capture ipu_bg_overlay_sdc ipu_still ipu_prp_enc ipu_csi_enc ipu_fg_overlay_sdc imx_vdoa ov5640_camera_mipi_int ov5640_camera_int v4l2_int_device caam secvio error evbug galcore(O)
[  127.423556] CPU: 0 PID: 70 Comm: ipu1_task Tainted: G           O      5.15.77-5.15.77-2.1.0+g9dcc132b0caa #1
[  127.433480] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[  127.440011] PC is at vdoa_get_handle+0x24/0xdc
[  127.444470] LR is at _get_vdoa_ipu_res+0x234/0x2c0
[  127.449279] pc : [<80991fe8>]    lr : [<8098bf54>]    psr: a00f0113
[  127.455552] sp : 822dbbf0  ip : 00000000  fp : 00000000
[  127.460780] r10: 82057a64  r9 : 82061d0c  r8 : 814a6fd4
[  127.466007] r7 : 860a9c00  r6 : 600f0113  r5 : 860a9dc4  r4 : 00000000
[  127.472537] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : 860a9dc4
[  127.479069] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  127.486211] Control: 10c5387d  Table: 163a404a  DAC: 00000051
[  127.491958] Register r0 information: slab kmalloc-512 start 860a9c00 pointer offset 452 size 512
[  127.500775] Register r1 information: NULL pointer
[  127.505488] Register r2 information: NULL pointer
[  127.510198] Register r3 information: NULL pointer
[  127.514907] Register r4 information: NULL pointer
[  127.519617] Register r5 information: slab kmalloc-512 start 860a9c00 pointer offset 452 size 512
[  127.528425] Register r6 information: non-paged memory
[  127.533483] Register r7 information: slab kmalloc-512 start 860a9c00 pointer offset 0 size 512
[  127.542115] Register r8 information: non-slab/vmalloc memory
[  127.547781] Register r9 information: non-slab/vmalloc memory
[  127.553448] Register r10 information: slab kmalloc-64 start 82057a40 pointer offset 36 size 64
[  127.562083] Register r11 information: NULL pointer
[  127.566880] Register r12 information: NULL pointer
[  127.571677] Process ipu1_task (pid: 70, stack limit = 0x1d76fa76)
[  127.577777] Stack: (0x822dbbf0 to 0x822dc000)
[  127.582141] bbe0:                                     00000032 81279780 00000001 814a6fd4
[  127.590326] bc00: 600f0113 8098bf54 ab61b7f0 80c91b0c 82286fb8 00000000 822dbc8c 860a9c00
[  127.598511] bc20: 814a6fd4 600f0113 8098fc74 814a7004 82061d0c 82057a64 00000000 8098da94
[  127.606695] bc40: 80c91b0c 82286fb8 822dbc8c 80c916d4 822dbc84 8015e460 822dbc64 00000031
[  127.614880] bc60: 00000046 846b73f9 00000001 822da000 82286c00 846b73f9 8098fc74 00000000
[  127.623066] bc80: 860a9c00 600f0113 8098fc74 814a7004 82061d0c 82057a64 00000000 8098fdd8
[  127.631249] bca0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  127.639433] bcc0: 81431b30 00000000 00000000 814a6fd4 00000000 00000000 00000002 1624e455
[  127.647618] bce0: 0530e455 062cf496 00000031 00000000 82286c00 8016ef88 822dbcf8 822dbcf8
[  127.655802] bd00: 81303d01 00000000 00000000 00000000 000014d3 00000400 000001e7 0000024c
[  127.663984] bd20: 00000000 00000000 00000000 00000001 00000000 00000000 00000000 846b73f9
[  127.672168] bd40: 00000000 822dbe54 2a3ae000 ab627780 82032a80 822dbe94 822dbee8 00000002
[  127.680352] bd60: 00000001 80165344 00000000 00000000 82045ec0 00006b14 00000800 00000000
[  127.688534] bd80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  127.696718] bda0: ffffffff 00000000 00000000 00000000 00000000 00000000 00005641 00000400
[  127.704901] bdc0: 00000010 00000005 00000000 00000000 00000000 00000001 00000000 00000000
[  127.713085] bde0: 00000000 846b73f9 00000000 82045ec0 2a3ae000 ab627780 82032a80 822dbe94
[  127.721268] be00: 822dbee8 80165778 00000000 00000000 81279780 00000000 00000000 00000000
[  127.729452] be20: 00000000 00000000 82032a80 81279780 00000000 00000000 00000002 00000000
[  127.737637] be40: 00000001 ab6243f0 822dbee8 ab627780 ffff8b45 82032a80 00000000 00000000
[  127.745820] be60: 00000001 ab627780 82045ed4 00000000 00000002 00000000 ab6243f0 00000000
[  127.754004] be80: 00000000 00000020 00000000 00000002 00000000 822dbe94 822dbe94 0018f7ed
[  127.762190] bea0: 82286c80 ab627780 ffff8b45 82032a80 00000000 ab627e50 ab627780 ffff8ae1
[  127.770374] bec0: 00000000 801669ac 822dbee8 82286c80 00000000 0a2ac732 00000000 ab627e40
[  127.778557] bee0: 00000001 81303d00 00000001 846b73f9 ab627780 ab627780 822dbf40 82286c00
[  127.786742] bf00: 822da000 00000000 82286c00 801532f4 ab627780 82286c00 810b6364 822da000
[  127.794926] bf20: 80c91b0c 82286fb8 822dbf6c 80c916d4 00000000 820804b4 20000093 82061d0c
[  127.803110] bf40: 822dbf74 846b73f9 00000000 822da000 82286c00 822da000 8098fc74 814a7024
[  127.811294] bf60: 82061d0c 846b73f9 822dbf7c 8237f280 82057a40 822da000 8098fc74 814a7024
[  127.819479] bf80: 82061d0c 82057a64 00000000 80151c3c 8237f280 80151ac4 00000000 00000000
[  127.827663] bfa0: 00000000 00000000 00000000 80100150 00000000 00000000 00000000 00000000
[  127.835846] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  127.844029] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  127.852215] [<80991fe8>] (vdoa_get_handle) from [<8098bf54>] (_get_vdoa_ipu_res+0x234/0x2c0)
[  127.860676] [<8098bf54>] (_get_vdoa_ipu_res) from [<8098da94>] (get_res_do_task+0x34/0x660)
[  127.869048] [<8098da94>] (get_res_do_task) from [<8098fdd8>] (ipu_task_thread+0x164/0xc24)
[  127.877332] [<8098fdd8>] (ipu_task_thread) from [<80151c3c>] (kthread+0x178/0x198)
[  127.884930] [<80151c3c>] (kthread) from [<80100150>] (ret_from_fork+0x14/0x24)
[  127.892171] Exception stack(0x822dbfb0 to 0x822dbff8)
[  127.897229] bfa0:                                     00000000 00000000 00000000 00000000
[  127.905412] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  127.913594] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  127.920217] Code: e24dd008 0a00001f e3a03000 e5853000 (e5943000) 
[  127.926394] ---[ end trace cf7302a9fdb2b359 ]---
[  128.379932] mxc_ipu mxc_ipu: ERR: [0xebdd3032] no-0x10, timeout:1000ms!
[  128.386597] mxc_ipu mxc_ipu: ERR: no-0x10,ipu_queue_task err:-110
[  129.419950] mxc_ipu mxc_ipu: ERR: [0x1640aba8] no-0x20, timeout:1000ms!
[  129.426616] mxc_ipu mxc_ipu: ERR: no-0x20,ipu_queue_task err:-110
[  130.459959] mxc_ipu mxc_ipu: ERR: [0x57bdf36f] no-0x30, timeout:1000ms!
[  130.466626] mxc_ipu mxc_ipu: ERR: no-0x30,ipu_queue_task err:-110
```

Does anybody else see this same error? @fabioestevam ? @MaxKrummenacher ? Did you ever see this? Or is this perhaps a problem with my own hardware? But I see this on two separate imx6dl sabre boards, which admittedly are old, but the same hw failure on both boards would be weird.

MaxKrummenacher, 2022-12-19:
Hi
We plan to use mainline only for the 32 bit i.MX Toradex modules. So I don't have a device tree and customized 5.15.77 kernel for a i.MX 6 to test. 

Does it help to not select mainline coda driver and with it not select the mainline vdoa driver? E.g. make sure that CONFIG_VIDEO_CODA and CONFIG_VIDEO_IMX_VDOA are not in the kernel config.

dv1, 2022-12-20:
@MaxKrummenacher I explicitly turned off `CONFIG_VIDEO_CODA` (the VDOA option was already off), and confirmed this:

```
# zcat /proc/config.gz | grep "CODA\|VDOA"
# CONFIG_VIDEO_CODA is not set
# CONFIG_CODA_FS is not set
```

The error is still there. I will try next to use linux-imx instead of linux-fslc-imx as the kernel, and see if the VDOA works with that.

dv1, 2022-12-29:
After quite some time, I think I found the cause and the solution: It boils down to a typo in a devicetree file, which causes a duplicate sysfs filename creation attempt, which in turn causes a crash in the kernel that prevents iram pools from being created. The VDOA needs this pool, and since the pool is absent, it cannot function.

This begins right at boot time. The kernel log shows:

```
[    0.078397] sysfs: cannot create duplicate filename '/devices/platform/soc/900000.sram'
[    0.078415] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.15.77-5.15.77-2.1.0+g9dcc132b0caa #1
[    0.078432] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[    0.078453] [<8010f14c>] (unwind_backtrace) from [<8010a1ac>] (show_stack+0x10/0x14)
[    0.078489] [<8010a1ac>] (show_stack) from [<80c83354>] (dump_stack_lvl+0x40/0x4c)
[    0.078517] [<80c83354>] (dump_stack_lvl) from [<80300468>] (sysfs_warn_dup+0x54/0x60)
[    0.078549] [<80300468>] (sysfs_warn_dup) from [<803005a0>] (sysfs_create_dir_ns+0xe8/0xfc)
[    0.078572] [<803005a0>] (sysfs_create_dir_ns) from [<80527434>] (kobject_add_internal+0xa0/0x330)
[    0.078606] [<80527434>] (kobject_add_internal) from [<80527c0c>] (kobject_add+0x5c/0xcc)
[    0.078630] [<80527c0c>] (kobject_add) from [<80697150>] (device_add+0xf8/0x898)
[    0.078662] [<80697150>] (device_add) from [<80947e50>] (of_platform_device_create_pdata+0x90/0xc0)
[    0.078694] [<80947e50>] (of_platform_device_create_pdata) from [<80948028>] (of_platform_bus_create+0x19c/0x230)
[    0.078716] [<80948028>] (of_platform_bus_create) from [<80948074>] (of_platform_bus_create+0x1e8/0x230)
[    0.078736] [<80948074>] (of_platform_bus_create) from [<80948228>] (of_platform_populate+0x70/0xd4)
[    0.078758] [<80948228>] (of_platform_populate) from [<8120bb00>] (imx6q_init_machine+0x98/0x2dc)
[    0.078785] [<8120bb00>] (imx6q_init_machine) from [<81203bc0>] (customize_machine+0x20/0x30)
[    0.078809] [<81203bc0>] (customize_machine) from [<801017b8>] (do_one_initcall+0x54/0x1d0)
[    0.078835] [<801017b8>] (do_one_initcall) from [<81201344>] (kernel_init_freeable+0x1b4/0x218)
[    0.078859] [<81201344>] (kernel_init_freeable) from [<80c900a4>] (kernel_init+0x10/0x128)
[    0.078888] [<80c900a4>] (kernel_init) from [<80100150>] (ret_from_fork+0x14/0x24)
```

The cause of the duplicate sysfs filename can be found in the `linux-fslc-imx` kernel in `arch/arm/boot/dts/imx6dl.dtsi`, where an `ocram` node uses an incorrect sram address that overlaps with what an `ocrams` node in `arch/arm/boot/dts/imx6qdl.dtsi` is already using.

Here's the node from `arch/arm/boot/dts/imx6dl.dtsi`:

```
		ocram: sram@900000 {
			compatible = "mmio-sram";
			reg = <0x00900000 0x20000>;
			ranges = <0 0x00900000 0x20000>;
			#address-cells = <1>;
			#size-cells = <1>;
			clocks = <&clks IMX6QDL_CLK_OCRAM>;
		};
```

Here's the node from `arch/arm/boot/dts/imx6qdl.dtsi`:

```
		ocrams: sram@00900000 {
			compatible = "fsl,lpm-sram";
			reg = <0x00900000 0x4000>;
			clocks = <&clks IMX6QDL_CLK_OCRAM>;
		};
```

It turns out that apparently there was a mixup during a merge. Address 0x900000 was already corrected to 0x905000 by a previous commit, but a different commit introduced an extra property to the `ocram` node, and accidentally reverted the address back to 0x900000.

I attached a patch for this so you can evaluate it. Let me know if it looks OK to you. The commit message contains these details as well.
[0001-ARM-dts-imx6dl-fix-reg-and-ranges-start-addresses.patch.gz](https://github.com/Freescale/meta-freescale/files/10322463/0001-ARM-dts-imx6dl-fix-reg-and-ranges-start-addresses.patch.gz)

otavio, 2022-12-30:
Could you prepare a PR? so we review it on the PR itself.

BTW, thanks for looking on this.

dv1, 2022-12-30:
[Done.](https://github.com/Freescale/linux-fslc/pull/617)

otavio, 2023-01-10:
Can I close this?

dv1, 2023-01-10:
Yeah, we can continue fixing this in the PR.

otavio, 2023-01-10:
PR has been merged.

closed 2023-01-10 by otavio