Freescale/meta-freescale #1353: Error: OpenCL error CL_INVALID_WORK_GROUP_SIZE (-54) during cv::Canny call

Christian-Prather, 2023-01-02:
### Description
Hello, I am currently working on a project that requires OpenCV GPU processing on the **imx8mq** processor. Currently I have OpenCL finding the GPU on the system and able to get device info, however when running a test program which calls cv::Canny() on the GPU I receive the following error
**_OpenCL error CL_INVALID_WORK_GROUP_SIZE (-54) during call: clEnqueueNDRangeKernel('row_filter_C1_D0', dims=2, globalsize=80x192x1, localsize=16x16x1) sync=false_** 

From what I can tell this error should be caused do to a mismatch in globalsize and localsize however these seem to follow the rules specified in the OpenCL docs. 

I am at a loss and would appreciate any support on this so I can utilize the GPU. I am using meta-freescale kirkstone opencv package.

### Error
OpenCL error CL_INVALID_WORK_GROUP_SIZE (-54) during call: clEnqueueNDRangeKernel('row_filter_C1_D0', dims=2, globalsize=80x192x1, localsize=16x16x1) sync=false

### Board
imx8mq

### Code to Reproduce
    cv::UMat normalizedImg, cannyImg, filledUMat;
    filledImg.copyTo(filledUMat);
    filledUMat.convertTo(normalizedImg, CV_8UC1, 1 / 256.0);
    cv::Canny(normalizedImg, cannyImg, cannyLowThresh, cannyHighThresh, cannyAperture, true);

### clinfo Results
Number of platforms                               1
  Platform Name                                   Vivante OpenCL Platform
  Platform Vendor                                 Vivante Corporation
  Platform Version                                OpenCL 3.0 V6.4.3.p4.398061
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_external_memory_dma_buf
  Platform Extensions with Version                cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
                                                  cl_khr_fp16                                                      0x400000 (1.0.0)
                                                  cl_khr_external_memory_dma_buf                                   0x400000 (1.0.0)
  Platform Numeric Version                        0xc00000 (3.0.0)
  Platform Host timer resolution                  0ns

  Platform Name                                   Vivante OpenCL Platform
Number of devices                                 1
  Device Name                                     Vivante OpenCL Device GC7000L.6214.0000
  Device Vendor                                   Vivante Corporation
  Device Vendor ID                                0x564956
  Device Version                                  OpenCL 3.0 
  Device Numeric Version                          0xc00000 (3.0.0)
  Driver Version                                  OpenCL 3.0 V6.4.3.p4.398061
  Device OpenCL C Version                         OpenCL C 1.2 
  Device OpenCL C all versions                    OpenCL C                                                         0x400000 (1.0.0)
                                                  OpenCL C                                                         0x401000 (1.1.0)
                                                  OpenCL C                                                         0x402000 (1.2.0)
                                                  OpenCL C                                                         0xc00000 (3.0.0)
  Device OpenCL C features                        __opencl_c_images                                                0x400000 (1.0.0)
                                                  __opencl_c_int64                                                 0x400000 (1.0.0)
  Latest conformance test passed                  v2021-03-25-00
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               1
  Max clock frequency                             800MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             1024x1024x1024
  Max work group size                             1024
  Preferred work group size multiple (device)     16
  Preferred work group size multiple (kernel)     16
  Max sub-groups per work group                   0
  Preferred / native vector sizes                 
    char                                                 4 / 4       
    short                                                4 / 4       
    int                                                  4 / 4       
[... 73 lines cut ...]
    Preferred size                                0
    Max size                                      0
  Max queues on device                            0
  Max events on device                            0
  Prefer user sync for interop                    Yes
  Profiling timer resolution                      1000ns
  Execution capabilities                          
    Run OpenCL kernels                            Yes
    Run native kernels                            No
    Non-uniform work-groups                       No
    Work-group collective functions               No
    Sub-group independent forward progress        No
    IL version                                    (n/a)
    ILs with version                              (n/a)
  printf() buffer size                            1048576 (1024KiB)
  Built-in kernels                                (n/a)
  Built-in kernels with version                   (n/a)
  Device Extensions                               cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_gl_sharing 
  Device Extensions with Version                  cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
                                                  cl_khr_fp16                                                      0x400000 (1.0.0)
                                                  cl_khr_global_int32_base_atomics                                 0x400000 (1.0.0)
                                                  cl_khr_global_int32_extended_atomics                             0x400000 (1.0.0)
                                                  cl_khr_local_int32_base_atomics                                  0x400000 (1.0.0)
                                                  cl_khr_local_int32_extended_atomics                              0x400000 (1.0.0)
                                                  cl_khr_gl_sharing                                                0x400000 (1.0.0)

otavio, 2023-01-12:
News @thochstein ?

thochstein, 2023-01-12:
Thanks for the reminder, this issue did get buried in my list...

I just made sure we didn't have any update to `opencv`, and it does seem up to date.

I'll ask the team here from some help on this.

thochstein, 2023-01-12:
@Christian-Prather, one question, are you on `master` or some other branch?

Christian-Prather, 2023-01-12:
@thochstein I am on the Kirkstone branch, thank you in advance for any help on this as well its very much appreciated!

Christian-Prather, 2023-02-09:
@thochstein Has there been any update on this?

thochstein, 2023-02-10:
Sorry, I have not heard. I will ping again my contacts.

Christian-Prather, 2023-02-13:
@thochstein  Thank you

thochstein, 2023-02-14:
Hi Christian,

Please do this: `export VIV_DEBUG=-MSG_LEVEL:WARNING`. Then ask your question here: https://community.nxp.com/

otavio, 2023-10-12:
Based on the absence of activity since February, I assume that the issue has been resolved.

closed 2023-10-12 by otavio