Freescale/meta-freescale #1377: u-boot-fslc fails to build for imx8mn_ddr4_evk after upgrade to 2023.01+fslc

zandrey, 2023-01-09:
Latest _U-Boot_ fails to build for `imx8mn_ddr4_evk` machine after upgrade has been performed via commit 65a85cfc9da2ec8872e3b431db84acc6f6ca6da9 with following output:
```
|   /storage/projects/mainline/build-output/work/imx8mn_ddr4_evk-linux/u-boot-fslc/2023.01+gitAUTOINC+386ee4eecc-r0/git/tools/binman/binman   --toolpath ./tools  build -u -d u-boot.dtb -O . -m  -I . -I /storage/projects/mainline/build-output/work/imx8mn_ddr4_evk-linux/u-boot-fslc/2023.01+gitAUTOINC+386ee4eecc-r0/git -I /storage/projects/mainline/build-output/work/imx8mn_ddr4_evk-linux/u-boot-fslc/2023.01+gitAUTOINC+386ee4eecc-r0/git/board/freescale/imx8mn_evk -I arch/arm/dts -a of-list="imx8mn-ddr4-evk"  -a atf-bl31-path= -a tee-os-path= -a opensbi-path= -a default-dt="imx8mn-ddr4-evk" -a scp-path= -a spl-bss-pad= -a tpl-bss-pad=1 -a spl-dtb=y -a tpl-dtb= -a pre-load-key-path=
| binman: Filename 'ddr4_imem_1d.bin' not found in input path (.,/storage/projects/mainline/build-output/work/imx8mn_ddr4_evk-linux/u-boot-fslc/2023.01+gitAUTOINC+386ee4eecc-r0/git,/storage/projects/mainline/build-output/work/imx8mn_ddr4_evk-linux/u-boot-fslc/2023.01+gitAUTOINC+386ee4eecc-r0/git/board/freescale/imx8mn_evk,arch/arm/dts) (cwd='/storage/projects/mainline/build-output/work/imx8mn_ddr4_evk-linux/u-boot-fslc/2023.01+gitAUTOINC+386ee4eecc-r0/build/imx8mn_ddr4_evk_defconfig')
```

While strictly not related to that commit per-se, latest _U-Boot_ version has changed the logic how the DDR4 firmware files are included by `binman` in commits [`93c4c0e4dd1e ("arm: dts: imx8mn-u-boot: Create common imx8mn-u-boot.dtsi")`](https://source.denx.de/u-boot/u-boot/-/commit/93c4c0e4dd1e75449fd48fc629823629d0c9bb58) and [`5a97f33938cf ("arm: dts: imx8mn-evk: Consolidate with imx8mn-u-boot")`](https://source.denx.de/u-boot/u-boot/-/commit/5a97f33938cf0c62dd36546cc8e4fc26d4c223a1), where firmware versions are not expected anymore. This causes `binman` fails to locate DDR4 firmware files, which leads to the build error.

Simple fix here would've been to drop the `DDR_FIRMWARE_VERSION` from the machine: https://github.com/Freescale/meta-freescale/blob/10c960f0eaa5fb774a676707c5148e29a6ae09b3/conf/machine/imx8mn-ddr4-evk.conf#L17

But looking at the Firmware package I saw that there are at least 3 DDR4 firmware versions present, all of those have different size and at that moment I decided to rather open this discussion to clarify the best course of action, specifically:
1. Shall the `DDR_FIRMWARE_VERSION` be dropped, and `ddr4_imem_1d.bin` file (and all other unversioned companions) be used to construct the `flash.bin`?
2. Or rather, `ddr4_dmem_1d_201810.bin` be deployed with renaming to `ddr4_dmem_1d.bin` instead?

@thochstein Is it safe to use unversioned DDR4 firmware files for this machine, or shall versioned ones be used instead?

Thanks a lot!

-- andrey

MrCry0, 2023-01-10:
@zandrey Thanks, I've reproduced this issue in a clean environment. @thochstein could you suggest the best for handling it?

thochstein, 2023-01-10:
I will forward to the team.

zandrey, 2023-01-10:
FTR, I've dropped the `DDR_FIRMWARE_VERSION` from the `imx8mn_ddr4_evk` machine and built the _core-image-minimal_ image - it had booted without any issues.

I've also tried the upstream _U-Boot_ standalone build (which does pull unversioned DDR FW files) - that one was also 100% operable.

I would just wait for final confirmation from @thochstein and the team, and if it turns out to be OK - drop the `DDR_FIRMWARE_VERSION` from the machine config.

-- andrey

thochstein, 2023-01-11:
Here's the response I got:
> Regarding to the FW version, we were told that a newer version supposes be able to work. But this never be tested and there is no request to upgrade the FW. So each platform still prefers using the FW version at the project beginning. At least this FW version has been tested by many release cycles.
> Therefore, I suggest to rename the ddr4_dmem_1d_201810.bin to ddr4_dmem_1d.bin, not using the one from 8MM project.
>

MrCry0, 2023-01-16:
@thochstein @zandrey @otavio well, I think the root of the issue is in the commit https://source.denx.de/u-boot/u-boot/-/commit/93c4c0e4dd1e75449fd48fc629823629d0c9bb58

Some `imx8mn-ddr4-*` machines had used versioned ddr4 fw, i.e. `imx8mn-ddr4-evk`. Some - unversioned, i.e `imx8mn-var-som-symphony`. But the common `imx8mn-u-boot.dtsi` has unversioned DDR4 FW only.

As NXP tested imx8mn with `ddr4_*_201810.bin` only,  I'd instead fix the file names of ddr4 fw in upstream `imx8mn-u-boot.dtsi`, using a backported patch for `meta-freescale` until the patch is merged and becomes available in `u-boot-fslc`.

otavio, 2023-01-16:
Agreed. We ought to use what NXP tests to avoid confusion. @fabioestevam as you did this rework in upstream what is your call on this?

fabioestevam, 2023-01-16:
It was Adam Ford that did the rework, not me.

I suggest that someone send a patch to the U-Boot list to rename the DDR4 firmware names in mx8mn-u-boot.dtsi.

otavio, 2023-01-16:
@MrCry0 can you send this upstream?

MrCry0, 2023-01-16:
> @MrCry0 can you send this upstream?

@otavio Done: https://patchwork.ozlabs.org/project/uboot/patch/20230116152127.1193051-1-oleksandr.suvorov@foundries.io/

closed 2023-01-16 by otavio