Freescale/meta-freescale #1399: Kirkstone: U-Boot: missing dependency on gnutls-native

jaskij, 2023-01-25:
When trying to build with the `imx8mp_rsb3720a1` config, I have encountered an error about missing `<gnutls/gnutls.h>`.

To quote from [this commit](https://patchwork.yoctoproject.org/project/arm/patch/20220928145346.48077-1-davidson.kumaresan@arm.com/),

> Since U-Boot 2022.04 the host tool mkeficapsule requires gnutls.

Adding a dependency on `gnutls-native` in a bbappend indeed solves the problem.

Sorry about not making, but I'm currently at work and we don't have a FOSS upstreaming policy. And I'd forget by the time I got home.

jaskij, 2023-01-25:
Thinking on it further, *why* this tool is even built for that config is probably worth digging into - or does U-Boot build all the host tool regardless of what the target needs?

jaskij, 2023-01-25:
Looking into it, the default configuration for this device does enable EFI, so at least my last comment is explained.

MrCry0, 2023-01-25:
Thanks, @jaskij, I'm working on this issue.

jaskij, 2023-03-23:
I actually did a fresh pull today, so I can test building, but sometime in the past year boots for this device broke (I'm in the middle of investigating it, actually), so I can't test booting.

otavio, 2023-03-23:
I'll close the issue then. Please open a new issue if the booting issue is caused by this layer.

closed 2023-03-23 by otavio

jaskij, 2023-03-24:
Unrelated to this issue, but as an FYI

> I'll close the issue then. Please open a new issue if the booting issue is caused by this layer.

I did some more testing, and the issue turned out to be in my BSP layer, not in meta-freescale.