Freescale/meta-freescale #1405: chromium patches do not apply and break build

ensc, 2023-02-03:
The shipped chromium patches (only kirkstone tested), do not apply on recent meta-browser layer (which ships chromium-109.0.5414.74).

I am not sure for which version they are; perhaps it would make sense to put them in a versioned .bbappend which can be `BBMASK`ed easily.

otavio, 2023-02-03:
@chrisdimich could you look at the patches issue?

chrisdimich, 2023-02-03:
Indeed, these patches are meant to apply to version 101.0.4951.54, as per https://github.com/nxp-imx/meta-imx/commits/kirkstone-5.15.71-2.2.0/meta-sdk/dynamic-layers/chromium-browser-layer/recipes-browser/chromium

@otavio want me to make them a versioned bbappend?

otavio, 2023-02-03:
It would be good to try to upgrade the patches.

chrisdimich, 2023-02-07:
Understood, but we only intended to mirror what NXP did and version 101.0.4951.54 combined with their patches works perfect for the 5.15.71-2.2.0 BSP. I would not be able to update for the latest. Can I please just version match the bbappend?

otavio, 2023-02-07:
This is exactly the kind of situation we shouldn't have. As your I'm not intending to keep maintaining the patches, it is better to just remove them.

We can't provide patches that are not maintained as that would be a security risk. I don't feel comfortable of providing support for something that is not maintained.

@

chrisdimich, 2023-02-08:
Okay understood. Would you like me to remove then?

otavio, 2023-02-08:
Yes, please prepare a PR removing this except if @thochstein, is able to upgrade the patches?

thochstein, 2023-02-08:
We're not able to manage frequent upgrades to these patches, so please go ahead and drop them.

chrisdimich, 2023-02-08:
Okay will do.

otavio, 2023-04-12:
Fixed in https://github.com/Freescale/meta-freescale/pull/1416.

closed 2023-04-12 by otavio