Freescale/meta-freescale #1406: Where to file issues against linux-fslc?

otavio, 2023-02-03:
### Discussed in https://github.com/Freescale/meta-freescale/discussions/1404

<div type='discussions-op-text'>

<sup>Originally posted by **jaskij** February  2, 2023</sup>
I'm not sure where to go - [linux-fslc](https://github.com/Freescale/linux-fslc/) has issues and discussions disabled.

I have found an issue specific to `linux-fslc` (not present in same version mainline) and have no idea where to send information about it.

---

Noting the info down so it doesn't get lost:

`add_one_port()` is called twice in `lpuart_probe()`, here:

https://github.com/Freescale/linux-fslc/blob/3ff5eb3ff57e665c47c072284d7f624e5452b85d/drivers/tty/serial/fsl_lpuart.c#L2965-L2986

Removing the second call seems to fix the issue (system has booted with console past the point, but I have not done extensive testing).

```patch
diff --git a/drivers/tty/serial/fsl_lpuart.c b/drivers/tty/serial/fsl_lpuart.c
index a12ff4954793..75e906c467d8 100644
--- a/drivers/tty/serial/fsl_lpuart.c
+++ b/drivers/tty/serial/fsl_lpuart.c
@@ -2981,10 +2981,6 @@ static int lpuart_probe(struct platform_device *pdev)
 	    sport->port.rs485.delay_rts_after_send)
 		dev_err(&pdev->dev, "driver doesn't support RTS delays\n");
 
-	ret = uart_add_one_port(&lpuart_reg, &sport->port);
-	if (ret)
-		goto failed_attach_port;
-
 	ret = devm_request_irq(&pdev->dev, sport->port.irq, handler, 0,
 				DRIVER_NAME, sport);
 	if (ret)
```</div>

otavio, 2023-02-03:
@jaskij, which kernel version is this?

@angolini did the last kernel stable release merges.

jaskij, 2023-02-03:
Ah, sorry. 5.15 as present in Kirkstone.

jaskij, 2023-02-03:
@otavio does this mean that meta-freescale is the correct place for issues specific to linux-fslc?

otavio, 2023-02-03:
No, it is `https://github.com/Freescale/linux-fslc`. Are you keen to open a PR?

otavio, 2023-02-03:
@jaskij, to avoid confusion. The preferred place to report issues is https://github.com/Freescale/linux-fslc as well as PRs.

That said, all contribution is welcome 👍🏼

jaskij, 2023-02-03:
@otavio the problem is, I can not create an issue in that repo - the tab is not shown to me at all. That's why I started a discussion here.

otavio, 2023-02-03:
Ouch! Fixed.

MrCry0, 2023-02-03:
https://github.com/Freescale/linux-fslc/pull/621

otavio, 2023-02-03:
@MrCry0 merged. Can you bump the kernel?

MrCry0, 2023-02-03:
@otavio sure! done: https://github.com/Freescale/meta-freescale/pull/1408

otavio, 2023-02-03:
Merged.

closed 2023-02-03 by otavio