Freescale/meta-freescale #1415: gstreamer1.0-plugins-bad: do not build on kirkstone

munoz0raul, 2023-02-08:
Hi All,

As commented on this PR: https://github.com/Freescale/meta-freescale/pull/1358

I have tested the patch that selects gcc for gstreamer1.0-plugins-bad on kirkstone and didn't work for me.

I also tried to revert that patch back to clang and also didn't work.

Following the logs:

munoz0raul, 2023-02-08:
```
bitbake-getvar -r gstreamer1.0-plugins-bad TOOLCHAIN
TOOLCHAIN="clang"
```

```
builder@8f1d20c0c480:/build/lmp/build-lmp-xwayland$ bitbake gstreamer1.0-plugins-bad
Loaded 5454 entries from dependency cache.
Parsing of 3715 .bb files complete (3714 cached, 1 parsed). 5455 targets, 683 skipped, 9 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-lmp-linux"
MACHINE              = "imx8mm-lpddr4-evk"
DISTRO               = "lmp-xwayland"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta-lmp-base        = "HEAD:1dbdf3f0870fbab1179265a48a18d0568ad7718c"
meta-oe              
meta-networking      
meta-filesystems     
meta-perl            
meta-python          = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-virtualization  = "HEAD:e68bfe7dae68326e8be64ebb3d99a1849c466e80"
meta-clang           = "kirkstone:e33eec34a85bac111efbf034dfe782fe3c105b05"
meta-updater         = "HEAD:ffbbc2b9f46a60bd2d51be53079252f8e6304a8b"
meta-security        
meta-tpm             
meta-parsec          
meta-integrity       = "HEAD:c79262a30bd385f5dbb009ef8704a1a01644528e"
meta-subscriber-overrides = "HEAD:e7dd0bf8a3ed75620a1d9ee05d65360d81c52a14"
meta-arm             
meta-arm-toolchain   
meta-arm-bsp         = "HEAD:67578fcfcd8ee8efcaef67ed7db1dfd55105872e"
meta-freescale       = "kirkstone:a609f6357c4fc3262445fffd8f2abf7130aebdb5"
meta-freescale-3rdparty = "HEAD:e1ec96f0b1d89adab9ec1f224e7d6dcb0ef565c0"
meta-raspberrypi     = "HEAD:dacad9302a92b0b7edf8546cdcad1f8ef753e462"
meta-intel           = "HEAD:f529e0594a784546926e89ce8e78385e00d0b0a9"
meta-yocto-bsp       = "HEAD:8f216dca2fda2c8862f81dacdcf11df3118b31b7"
meta-xilinx-core     
meta-xilinx-bsp      
meta-xilinx-standalone = "HEAD:f8d8efab12040712df32436643621b2756de1f76"
meta-xilinx-tools    = "HEAD:9acf9d1d02f465b3c435283f92557b632becc72a"
meta-tegra           = "HEAD:9d58efcfad33a0496b2099e01111c4c937bf3287"
meta-ti-bsp          = "HEAD:2124e7ecd88d1aded320e86da62785c3ab171b27"
meta-st-stm32mp      = "HEAD:00c2494b66b8fa2c0fc2008fa802a2adbeed966e"
meta-sunxi           = "HEAD:773dc04614a4b03c1572cf4b778649c506051176"
meta-lmp-bsp         = "HEAD:1dbdf3f0870fbab1179265a48a18d0568ad7718c"
meta-wayland         = "HEAD:3e32a764d25e844819bfc44b82f8d13c611be954"
meta-qt5             = "master:1d1b19ff577835bf847152eed44d52e8267d9093"
meta                 = "HEAD:5b2ee67e3a5587b4c7d97d2a9bc00022d1eedae3"

Initialising tasks: 100% |##############################################################################################################################################################################################| Time: 0:00:06
Sstate summary: Wanted 576 Local 540 Mirrors 0 Missed 36 Current 1988 (93% match, 98% complete)
NOTE: Executing Tasks
ERROR: gstreamer1.0-plugins-bad-1.20.3.imx-r0 do_compile: ExecutionError('/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/temp/run.do_compile.195410', 1, None, None)
ERROR: Logfile of failure stored in: /build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/temp/log.do_compile.195410
Log data follows:
| DEBUG: Executing shell function do_compile
| [1/882] aarch64-lmp-linux-clang -target aarch64-lmp-linux -mcpu=cortex-a53 -march=armv8-a+crc+crypto -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/recipe-sysroot -Igst-libs/gst/basecamerabinsrc/libgstbasecamerabinsrc-1.0.so.0.2003.0.p -Igst-libs/gst/basecamerabinsrc -I../git/gst-libs/gst/basecamerabinsrc -I. -I../git -Igst-libs -I../git/gst-libs -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/recipe-sysroot/usr/include/gstreamer-1.0 -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/recipe-sysroot/usr/include/glib-2.0 -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/recipe-sysroot/usr/lib/glib-2.0/include -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0/recipe-sysroot/usr/include/imx -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O0 -fvisibility=hidden -fno-strict-aliasing -DG_DISABLE_CAST_CHECKS -Wmissing-prototypes -Wdeclaration-after-statement -Wold-style-definition -Wmissing-declarations -Wredundant-decls -Wwrite-strings -Wformat -Wformat-security -Winit-self -Wmissing-include-dirs -Waddress -Wno-multichar -Wvla -Wpointer-arith -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0=/usr/src/debug/gstreamer1.0-plugins-bad/1.20.3.imx-r0 -fdebug-prefix-map=/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.3.imx-r0=/usr/src/debug/gs [... cut ...]
ERROR: Task (/build/lmp/build-lmp-xwayland/conf/../../layers/meta-freescale/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.20.3.imx.bb:do_compile) failed with exit code '1'

Second Keyboard Interrupt, stopping...

Summary: 1 task failed:
  /build/lmp/build-lmp-xwayland/conf/../../layers/meta-freescale/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.20.3.imx.bb:do_compile
Summary: There was 1 ERROR message, returning a non-zero exit code.
builder@8f1d20c0c480:/build/lmp/build-lmp-xwayland$ 
```

munoz0raul, 2023-02-08:
Using GCC:
```
bitbake-getvar -r gstreamer1.0-plugins-bad TOOLCHAIN
TOOLCHAIN="gcc"
```

```
bitbake lmp-factory-image
NOTE: Reconnecting to bitbake server...
Loading cache: 100% |########################################################################################################################################| Time: 0:00:01
Loaded 5453 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################| Time: 0:00:02
Parsing of 3714 .bb files complete (3712 cached, 2 parsed). 5454 targets, 682 skipped, 9 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-lmp-linux"
MACHINE              = "imx8mm-lpddr4-evk"
DISTRO               = "lmp-xwayland"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta-lmp-base        = "HEAD:1dbdf3f0870fbab1179265a48a18d0568ad7718c"
meta-oe              
meta-networking      
meta-filesystems     
meta-perl            
meta-python          = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-virtualization  = "HEAD:e68bfe7dae68326e8be64ebb3d99a1849c466e80"
meta-clang           = "HEAD:013b5eeb580bc13b156aaf77d0c1175f75b89671"
meta-updater         = "HEAD:ffbbc2b9f46a60bd2d51be53079252f8e6304a8b"
meta-security        
meta-tpm             
meta-parsec          
meta-integrity       = "HEAD:c79262a30bd385f5dbb009ef8704a1a01644528e"
meta-subscriber-overrides = "HEAD:e7dd0bf8a3ed75620a1d9ee05d65360d81c52a14"
meta-arm             
meta-arm-toolchain   
meta-arm-bsp         = "HEAD:67578fcfcd8ee8efcaef67ed7db1dfd55105872e"
meta-freescale       = "HEAD:cea64da30c79a56517abe2d125b295a6e248680c"
meta-freescale-3rdparty = "HEAD:e1ec96f0b1d89adab9ec1f224e7d6dcb0ef565c0"
meta-raspberrypi     = "HEAD:dacad9302a92b0b7edf8546cdcad1f8ef753e462"
meta-intel           = "HEAD:f529e0594a784546926e89ce8e78385e00d0b0a9"
meta-yocto-bsp       = "HEAD:8f216dca2fda2c8862f81dacdcf11df3118b31b7"
meta-xilinx-core     
meta-xilinx-bsp      
meta-xilinx-standalone = "HEAD:f8d8efab12040712df32436643621b2756de1f76"
meta-xilinx-tools    = "HEAD:9acf9d1d02f465b3c435283f92557b632becc72a"
meta-tegra           = "HEAD:9d58efcfad33a0496b2099e01111c4c937bf3287"
meta-ti-bsp          = "HEAD:2124e7ecd88d1aded320e86da62785c3ab171b27"
meta-st-stm32mp      = "HEAD:00c2494b66b8fa2c0fc2008fa802a2adbeed966e"
meta-sunxi           = "HEAD:773dc04614a4b03c1572cf4b778649c506051176"
meta-lmp-bsp         = "HEAD:1dbdf3f0870fbab1179265a48a18d0568ad7718c"
meta-wayland         = "HEAD:3e32a764d25e844819bfc44b82f8d13c611be954"
meta-qt5             = "master:1d1b19ff577835bf847152eed44d52e8267d9093"
meta                 = "HEAD:5b2ee67e3a5587b4c7d97d2a9bc00022d1eedae3"

Initialising tasks: 100% |###################################################################################################################################| Time: 0:00:09
Sstate summary: Wanted 1035 Local 874 Mirrors 0 Missed 161 Current 3619 (84% match, 96% complete)
NOTE: Executing Tasks
ERROR: gstreamer1.0-plugins-bad-1.20.0.imx-r0 do_compile: ExecutionError('/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/temp/run.do_compile.882172', 1, None, None)
ERROR: Logfile of failure stored in: /build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/temp/log.do_compile.882172
Log data follows:
| DEBUG: Executing shell function do_compile
| [1/882] aarch64-lmp-linux-gcc -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/recipe-sysroot -Igst-libs/gst/codecparsers/libgstcodecparsers-1.0.so.0.2000.0.p -Igst-libs/gst/codecparsers -I../git/gst-libs/gst/codecparsers -I. -I../git -Igst-libs -I../git/gst-libs -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/recipe-sysroot/usr/include/gstreamer-1.0 -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/recipe-sysroot/usr/include/glib-2.0 -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/recipe-sysroot/usr/lib/glib-2.0/include -I/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0/recipe-sysroot/usr/include/imx -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -fvisibility=hidden -fno-strict-aliasing -DG_DISABLE_CAST_CHECKS -Wmissing-prototypes -Wdeclaration-after-statement -Wold-style-definition -Wmissing-declarations -Wredundant-decls -Wwrite-strings -Wformat -Wformat-security -Winit-self -Wmissing-include-dirs -Waddress -Wno-multichar -Wvla -Wpointer-arith -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/build/lmp/build-lmp-xwayland/tmp-lmp_xwayland/work/cortexa53-crypto-mx8mm-lmp-linux/gstreamer1.0-plugins-bad/1.20.0.imx-r0=/usr/src/debug/gstreamer1.0-plugins-bad/ [... cut ...] r.c.o -MF gst-libs/gst/codecparsers/libgstcodecparsers-1.0.so.0.2000.0.p/gstvc1parser.c.o.d -o gst-libs/gst/codecparsers/libgstcodecparsers-1.0.so.0.2000.0.p/gstvc1parser.c.o -c ../git/gst-libs/gst/codecparsers/gstvc1parser.c
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/build/lmp/build-lmp-xwayland/conf/../../layers/meta-freescale/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.20.0.imx.bb:do_compile) failed with exit code '1'

```

otavio, 2023-02-08:
@thochstein, this seems to be an issue with the NXP fork. Can you check?

thochstein, 2023-02-08:
We have updated to 1.20.3 in our layer for `kirkstone`: https://github.com/nxp-imx/meta-imx/tree/kirkstone-5.15.71-2.2.0/meta-bsp/recipes-multimedia/gstreamer

otavio, 2023-02-08:
@munoz0raul can you work on a PR upgrading it?

munoz0raul, 2023-02-10:
Hello @otavio and @thochstein.

I just tryied with the gstreamer recipes from meta-imx and got the same error.

The solution I found was to use ``gcc`` for ``gobject-introspection`` as well. That fix the build issue. 

`TOOLCHAIN:pn-gobject-introspection:imx = "gcc"`

thochstein, 2023-02-10:
Good find!

FYI @kraj

kraj, 2023-02-10:
looks ok to me.

otavio, 2023-02-11:
This is interesting. However, I wonder why this doesn't happen in mainline `gstreamer1.0-plugins-bad`. Máybe we have a different `PACKAGECONFIG` setting?

munoz0raul, 2023-02-11:
This is my PACKAGECONFIG:
```
bitbake-getvar -r gstreamer1.0-plugins-bad PACKAGECONFIG
PACKAGECONFIG="     orc     bluez      x11     wayland     gl     bz2 closedcaption curl dash  hls openssl sbc smoothstreaming     sndfile ttml uvch264 webp     rsvg  kms tinycompress"
```

And these are the layers I have:

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-lmp-linux"
MACHINE              = "imx8mm-lpddr4-evk"
DISTRO               = "lmp-xwayland"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta-lmp-base        = "HEAD:1dbdf3f0870fbab1179265a48a18d0568ad7718c"
meta-oe              
meta-networking      
meta-filesystems     
meta-perl            
meta-python          = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-virtualization  = "HEAD:e68bfe7dae68326e8be64ebb3d99a1849c466e80"
meta-clang           = "HEAD:013b5eeb580bc13b156aaf77d0c1175f75b89671"
meta-updater         = "HEAD:ffbbc2b9f46a60bd2d51be53079252f8e6304a8b"
meta-security        
meta-tpm             
meta-parsec          
meta-integrity       = "HEAD:c79262a30bd385f5dbb009ef8704a1a01644528e"
meta-subscriber-overrides = "HEAD:46988d6e386ab8c017c79a98cf812ee82d41c87f"
meta-arm             
meta-arm-toolchain   
meta-arm-bsp         = "HEAD:67578fcfcd8ee8efcaef67ed7db1dfd55105872e"
meta-freescale       = "HEAD:cea64da30c79a56517abe2d125b295a6e248680c"
meta-freescale-3rdparty = "HEAD:e1ec96f0b1d89adab9ec1f224e7d6dcb0ef565c0"
meta-raspberrypi     = "HEAD:dacad9302a92b0b7edf8546cdcad1f8ef753e462"
meta-intel           = "HEAD:f529e0594a784546926e89ce8e78385e00d0b0a9"
meta-yocto-bsp       = "HEAD:8f216dca2fda2c8862f81dacdcf11df3118b31b7"
meta-xilinx-core     
meta-xilinx-bsp      
meta-xilinx-standalone = "HEAD:f8d8efab12040712df32436643621b2756de1f76"
meta-xilinx-tools    = "HEAD:9acf9d1d02f465b3c435283f92557b632becc72a"
meta-tegra           = "HEAD:9d58efcfad33a0496b2099e01111c4c937bf3287"
meta-ti-bsp          = "HEAD:2124e7ecd88d1aded320e86da62785c3ab171b27"
meta-st-stm32mp      = "HEAD:00c2494b66b8fa2c0fc2008fa802a2adbeed966e"
meta-sunxi           = "HEAD:773dc04614a4b03c1572cf4b778649c506051176"
meta-lmp-bsp         = "HEAD:1dbdf3f0870fbab1179265a48a18d0568ad7718c"
meta-webkit          = "HEAD:ff5ab70d54ad250f57328d6e7c2c37b9ceaf3748"
meta                 = "HEAD:5b2ee67e3a5587b4c7d97d2a9bc00022d1eedae3"

```

otavio, 2023-02-11:
So, here are a couple of steps to help to isolate the issue:

1. check if this happens in official recipes or if it is specific to NXP's fork
2. check if this happens in fslc-community-bsp

Once we know this, we can reduce the search scope for the root cause.

otavio, 2023-04-13:
@munoz0raul, did you find the root cause? Otherwise, we'll close this issue due to inactivity.

otavio, 2023-04-24:
Closing due to inactiveness.

closed 2023-04-24 by otavio