Freescale/meta-freescale #1426: Rework firmware-imx-8 recipe

otavio, 2023-02-13:
As discussed in #1414, we agreed to rework the recipe because `mx8` and `mx9` devices are using it. We need to decide the recipe name for use as it is used for boot blob generation and isn't intended to be installed inside the rootfs directly.

otavio, 2023-02-13:
My first idea for the recipe name is `imx-boot-firmware-files`.

thochstein, 2023-02-13:
The `SRC_URI` is not for `imx-boot` components only and include i.MX 6 support:

```
nxa21648@lsv07069:~/upstream/sources/meta-freescale/recipes-bsp/firmware-imx$ grep -r "require firmware-imx"
firmware-imx_8.18.bb:require firmware-imx-${PV}.inc
firmware-imx-8_8.18.bb:require firmware-imx-${PV}.inc
firmware-imx-8m_8.18.bb:require firmware-imx-${PV}.inc
```

otavio, 2023-02-13:
All non-imx-boot specific files should be installed on demand and use the packages. We can share the `SRC_URI` but likely have two recipes only.

commit 90df14f9 (2023-04-25): imx-boot-firmware-files: consolidate deploy firmware files across SoCs

The firmware-imx-8 and firmware-imx-8m recipes are mostly code
duplication and some parts even used by i.MX93 so consolidating it
allows code reuse and clearness.

Fixes: #1426.
closed 2023-04-25 by otavio