Freescale/meta-freescale #1438: Error build images using machines with qoriq linux

mdrodrigo, 2023-02-22:
We found an issue when building the images using machines with the qoriq Linux. 
The errors found were using that machines:
ls1012afrwy
ls1012ardb
ls1021atwr
ls1028ardb
ls1043ardb
ls1046afrwy
ls1046ardb
ls1088ardb
ls1088ardb-pb
ls2080ardb
ls2088ardb
lx2160ardb
lx2162aqds
mpc8548cds
p1020rdb
p2020rdb
p2041rdb
p3041ds
p4080ds
p5040ds-64b
p5040ds
t1024rdb-64b
t1024rdb
t1042d4rdb-64b
t1042d4rdb
t2080rdb-64b
t2080rdb
t4240rdb-64b
t4240rdb
The errors are with the optee-os-qoriq, optee-test-qoriq, optee-alf.
Some errors are related to machine incompatibility.

otavio, 2023-02-22:
@junzhuimx could you have a look?

junzhuimx, 2023-03-10:
@mdrodrigo  more details about build errors?

otavio, 2023-04-12:
@junzhuimx it seems something has changed in past and causes the machines to fail. Who is taking care of QorIQ?

otavio, 2023-04-12:
One failure example:

```
| aarch64-oel-linux-ld.bfd: warning: /home/otavio/src/oe/oel/master/build/tmp/work/ls1012afrwy-oel-linux/qoriq-atf/2.4-r0/git/build/ls1012afrwy/release/bl2/bl2.elf has a LOAD segment with RWX permissions
| make: *** [Makefile:1089: /home/otavio/src/oe/oel/master/build/tmp/work/ls1012afrwy-oel-linux/qoriq-atf/2.4-r0/git/build/ls1012afrwy/release/bl2/bl2.elf] Error 1
| make: *** Waiting for unfinished jobs....
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
```

otavio, 2023-04-13:
I'll report a new issue with updated failing list.

closed 2023-04-13 by otavio