Freescale/meta-freescale #1447: i.MX8 : Install fitImage in rootfs broken

embetrix, 2023-03-09:
when enabling fitImage build:

```
KERNEL_CLASSES    += "kernel-fitimage"
KERNEL_IMAGETYPES += "fitImage"
```

The fitImage is generated in DEPLOY directory but installation in rootfs there is only a broken symlink:

```
ls -l /boot/
total 29130
lrwxrwxrwx. 1 root root       44 Mär  9  2018 fitImage -> fitImage-5.15.67-5.15.60-2.0.0+g845d53f13aac
lrwxrwxrwx. 1 root root       41 Mär  9  2018 Image -> Image-5.15.67-5.15.60-2.0.0+g845d53f13aac
-rw-r--r--. 1 root root 31025664 Mär  9  2018 Image-5.15.67-5.15.60-2.0.0+g845d53f13aac
```

This is definitely related to **meta-freescale** layer since it's working for other builds, I'm using an i.MX8

otavio, 2023-03-09:
What is the machine you are using?

embetrix, 2023-03-09:
MACHINE=imx8mp-lpddr4-evk

otavio, 2023-04-12:
I couldn't reproduce this. Using my image, I have:
```console
% grep fit buildhistory/images/imx8mm_lpddr4_evk/glibc/core-image-base/files-in-image.txt
-rw-r--r-- root       root         12448560 ./boot/fitImage-6.1.20-fslc+g526cda8a8d92
lrwxrwxrwx root       root               34 ./boot/fitImage -> fitImage-6.1.20-fslc+g526cda8a8d92
```
Please double-check if you can still reproduce this. I tested the `master` branch.

embetrix, 2023-04-13:
The problem was caused by _INITRAMFS_IMAGE_BUNDLE = "1"_ enabled, of course this will not go to rootfs because of a cirular dependency.

closed 2023-04-13 by embetrix