Freescale/meta-freescale #1463: Upgrade to NXP BSP release L6.1.1-1.0.0

otavio, 2023-03-23:
The immediate items to work on are:

- [ ] Upgrade Linux kernel and U-Boot (i.MX variants)
- [ ] Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants)
- [ ] Upgrade OP-TEE and ATF (i.MX variants)
- [ ] Upgrade Multimedia Packages
- [ ] Upgrade Graphics Packages

**TODO**: ~The Release Notes are here for checking~.

angolini, 2023-03-23:
I think I can start thinking about this next week. I hope so

thochstein, 2023-03-31:
> * [ ]  Upgrade Linux kernel and U-Boot (i.MX variants)

Working on this.

MrCry0, 2023-04-02:
> * [ ]  Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants)

Working on this.

otavio, 2023-04-03:
I added the release notes link.

otavio, 2023-04-17:
@MrCry0 I upgraded the #1501 and I'd like to know what is missing for the FSLC fork of Linux.

MrCry0, 2023-04-18:
> @MrCry0 I upgraded the #1501 and I'd like to know what is missing for the FSLC fork of Linux.

@otavio I just made `6.1-1.0.x-imx` updated to the latest stable 6.1 built well. Today, I'm going to provide updates for other fslc parts.

otavio, 2023-04-18:
@MrCry0 can I review the PR?

MrCry0, 2023-04-18:
> @MrCry0 can I review the PR?

@otavio Sure, I've already made the PR ready :) https://github.com/Freescale/linux-fslc/pull/629

otavio, 2023-04-18:
@MrCry0 merged. Please prepare the PR.

MrCry0, 2023-04-18:
> @MrCry0 merged. Please prepare the PR.

Thanks! Sure, let me finish the BSP building.

MrCry0, 2023-04-19:
> @MrCry0 merged. Please prepare the PR.

Hi @otavio There is another small PR https://github.com/Freescale/linux-fslc/pull/630 for `linux-fslc-imx` kernel and with that merged, the PR https://github.com/Freescale/meta-freescale/pull/1504 will be ready to review as well.

MrCry0, 2023-04-24:
@otavio I can take the next OP-TEE + ATF task.

closed 2023-06-13 by otavio