Freescale/meta-freescale #1491: EULA clarification

Akbar-arm, 2023-04-11:
Appendix A in the EULA lists a large amount of third-party software from lots of third-party companies. It is unclear which third party software is actually used in this layer. Is there any way to retrieve this information? The only company I can see explicitly mentioned is Cortina in the custom-licenses directory. I am assuming the EULA is generic and lists more than what is included in this layer.

otavio, 2023-04-12:
The layer itself fetches sources from multiple places. Specifically about the NXP components, those use different licenses, and some include 3rd party IP. The [SCR-6.1.1-1.0.0.txt](https://github.com/Freescale/meta-freescale/blob/master/SCR/SCR-6.1.1-1.0.0.txt) provides a list for reference.

Akbar-arm, 2023-04-13:
Hi @otavio, I am still confused. The license information is extremely important but it is still not clear.

1. The EULA contains a large list of 3rd party IP that is not referenced in the SCR-6.1.1-1.0.0.txt file. Which file takes precedence? Am I correct in thinking the EULA contains a large list of 3rd party IP not included in this layer?
2. There is a custom-licenses directory containing a few licenses but again, it is not clear which recipes they apply to
3. In the SCR-6.1.1-1.0.0.txt file, the outgoing license mentions "Additional distribution license granted - Section 2.3 applies" but the package deepview-rt-2.4.46.1-aarch64.bin mentions "No distribution license granted - Section 2.2 applies". This seems like a mistake. Are we expected to remove this package to have the additional distribution license?

Given the issues above, I wonder whether it's possible to instead forgo this layer altogether. How essential is this layer? Are there alternatives to create a yocto linux image for the i.MX 8M Plus platform with a more permissive license?

ann-thornton, 2023-04-21:
The LA_OPT_NXP_Software_License is used in many other products than just this one so it includes special cases that may not apply here.  The SCR lists all 3rd party software that NXP contributes.  So if anything is listed in the SCR that is listed in the license appendices, the appendix conditions apply.  I know, it is a lot to go through in this case.  If it helps, you only need to look at the proprietary software listed in the SCR.  The SCR tells you which licenses apply to which components.  
Deepview-rt is an optional package that can easily be removed from the recipes.  It is not required and is there for demo purposes. The license is not a mistake.

Akbar-arm, 2023-04-21:
Thank you. What about the "custom-licenses" directory, how do those licenses apply and to what packages?
Also, presumably Deepview-rt is not enabled by default as otherwise, the outgoing license of the SCR file is misleading. I also can't see any mentions of "deepview" in the repository (other than the SCR) so I don't know how to check that it is not used/ which recipe it is in.

ann-thornton, 2023-04-21:
The NXP and Freescale licenses in custom-licenses look like they are old and have been replaced by the LA_OPT_NXP_Software_License.  They may be there because something older is still included somewhere. The others probably come from other contributors..

otavio, 2023-10-12:
Since there has been no activity since April, I assume that the issue has been resolved.

closed 2023-10-12 by otavio