Freescale/meta-freescale #1495: [qoriq] Build failure for qoriq-atf recipe version 2.4

otavio, 2023-04-13:
I encountered a build failure while attempting to build the `qoriq-atf` recipe version 2.4. The error message indicates that the build failed while attempting to compile the `bl2.elf` and `bl31.elf` files. The error message also contains the following warning related to `bl31.elf`: "`aarch64-oel-linux-ld.bfd: warning: [path_to_bl31.elf] has a LOAD segment with RWX permissions`".

The build fails on the following machines:

- [ ] `ls1012afrwy`
- [ ] `ls1012ardb`
- [ ] `ls1028ardb`
- [ ] `ls1043ardb`
- [ ] `ls1046afrwy`
- [ ] `ls1046ardb`
- [ ] `ls1088ardb`
- [ ] `ls1088ardb-pb`
- [ ] `ls2088ardb`
- [ ] `lx2160ardb`
- [ ] `lx2162aqds`

These machines have been broken for a long time and if no one is interested in maintaining them, it would be preferable to remove them from the repository.

This issue appears to be similar to the one fixed in https://github.com/Freescale/meta-freescale/pull/1179/commits/bac8c6af8471addb7add0d1b1ddfbbbd36bed16e.

Please advise on how to proceed with fixing the build for these machines or removing them from the repository.

junzhuimx, 2023-04-17:
It is the because of binutils upgrade to 2.39.
Let me fix it by upgrade to the latest ATF v2.6

otavio, 2023-04-17:
Fixed in #1500

otavio, 2023-04-17:
@junzhuimx the fix seems incomplete. Now I have:

```
ERROR: Nothing PROVIDES 'mbedtls' (but /home/otavio/src/oe/oel/master/sources/meta-freescale/recipes-bsp/atf/qoriq-atf_2.6.bb DEPENDS on or otherwise requires it)
ERROR: Required build target 'core-image-base' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-base', 'qoriq-atf', 'mbedtls']
```

junzhuimx, 2023-04-19:
@otavio raise another PR to fix it

otavio, 2023-04-19:
Fixed. Thanks @junzhuimx

closed 2023-04-19 by otavio