Freescale/meta-freescale #1496: [qoriq] Error building `u-boot-qoriq` for QorIQ machines - `ModuleNotFoundError: No module named 'pkg_resources'`

otavio, 2023-04-13:
I encountered a build failure while attempting to build the `u-boot-qoriq` recipe version 2021.04+fslgit-r0. This error seems to be caused by a missing module called `pkg_resources`. The build command involves using `binman` tool that calls this missing module, causing the build to fail.

The build fails on the following machines:

- [ ] `mpc8548cds`
- [ ] `p2041rdb`
- [ ] `p3041ds`
- [ ] `p4080ds`
- [ ] `p5040ds-64b`
- [ ] `t4240rdb-64b`

These machines have been broken for a long time and if no one is interested in maintaining them, it would be preferable to remove them from the repository.

Please advise on how to proceed with fixing the build for these machines or removing them from the repository.

otavio, 2023-04-24:
This is still reproducible. Are anyone going to look at this or should we remove the machines?

jeffkize, 2023-06-30:
I fixed it by updating the recipe (adding python3-setuptools-native):
DEPENDS = "libgcc virtual/${TARGET_PREFIX}gcc bison-native bc-native swig-native python3-native python3-setuptools-native"

otavio, 2023-10-12:
@jeffkize, would you be able to create a pull request that resolves this issue?

MrCry0, 2024-02-17:
@otavio does the 1741 fix this issue?

otavio, 2024-02-17:
Indeed; closing.

closed 2024-02-17 by otavio