Freescale/meta-freescale #1498: [qoriq] rcw-git-r0 do_compile: oe_runmake failed with "No such file or directory" error for `ls2080ardb`

otavio, 2023-04-13:
I encountered an error while building the `rcw` recipe. Specifically, the do_compile task for the rcw_git recipe failed with an error message indicating that oe_runmake failed with a "No such file or directory" error.

The error message itself reads:

```
ERROR: rcw-git-r0 do_compile: oe_runmake failed
ERROR: rcw-git-r0 do_compile: ExecutionError('/home/otavio/src/oe/oel/master/build/tmp/work/ls2080ardb-oel-linux/rcw/git-r0/temp/run.do_compile.2333091', 1, None, None)
ERROR: Logfile of failure stored in: /home/otavio/src/oe/oel/master/build/tmp/work/ls2080ardb-oel-linux/rcw/git-r0/temp/log.do_compile.2333091
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 24 BOARDS=ls2080ardb DESTDIR=/home/otavio/src/oe/oel/master/build/tmp/work/ls2080ardb-oel-linux/rcw/git-r0/image/boot/rcw/
| make[1]: *** ls2080ardb: No such file or directory. Stop.
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| make: *** [Makefile:18: all] Error 2
NOTE: recipe rcw-git-r0: task do_compile: Failed
ERROR: Task (/home/otavio/src/oe/oel/master/sources/meta-freescale/recipes-bsp/rcw/rcw_git.bb:do_compile) failed with exit code '1'
```

This suggests that a required file or directory is missing or misconfigured, specifically the `ls2080ardb`.

Any assistance or advice on resolving this issue would be greatly appreciated. Thank you!

junzhuimx, 2023-04-17:
@otavio Let me check with our develop team to determine whether ls2080 is supported or not

otavio, 2023-04-24:
This is still reproducible. Are anyone going to look at this or should we remove the machines?

junzhuimx, 2023-08-30:
@otavio we can drop ls2080 now

otavio, 2023-08-30:
Would you mind preparing the pull request?

junzhuimx, 2023-09-01:
@otavio Yes, I will raise a PR later

commit a8d8dc46 (2023-10-12): ls2080ardb: drop as it is unbuildable for too long

Fixes: #1498.
closed 2023-10-12 by otavio