Freescale/meta-freescale #1549: imx-atf does not build with LD from binutils 2.40 available in poky/mickledore

dwretman, 2023-05-26:
Building imx-atf from master 17a762639f1d3f355291a3de4767983deaf4493d using LD from binutils 2.40 available in poky/mickledore fails with:

```
|   LD      /build-cache/tmp-mickledore/work/fsimx8mn-poky-linux/imx-atf/2.6+gitAUTOINC+35f4c7295b-r0/git/build/imx8mn/release/bl31/bl31.elf
| aarch64-poky-linux-ld: warning: /build-cache/tmp-mickledore/work/fsimx8mn-poky-linux/imx-atf/2.6+gitAUTOINC+35f4c7295b-r0/git/build/imx8mn/release/bl31/bl31.elf has a LOAD segment with RWX permissions
| make: *** [Makefile:1306: /build-cache/tmp-mickledore/work/fsimx8mn-poky-linux/imx-atf/2.6+gitAUTOINC+35f4c7295b-r0/git/build/imx8mn/release/bl31/bl31.elf] Error 1
```

This issue has been fixed in atf upstream (https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/19401) and the build problem can be worked around with the following patch (modeled from the upstream change):

```diff
diff --git a/Makefile b/Makefile
index 3941f8698..6d5973013 100644
--- a/Makefile
+++ b/Makefile
@@ -409,6 +409,8 @@ endif
 
 GCC_V_OUTPUT           :=      $(shell $(CC) -v 2>&1)
 
+TF_LDFLAGS             +=      -z noexecstack
+
 # LD = armlink
 ifneq ($(findstring armlink,$(notdir $(LD))),)
 TF_LDFLAGS             +=      --diag_error=warning --lto_level=O1
@@ -435,6 +437,9 @@ TF_LDFLAGS          +=      $(subst --,-Xlinker --,$(TF_LDFLAGS_$(ARCH)))
 
 # LD = gcc-ld (ld) or llvm-ld (ld.lld) or other
 else
+# With ld.bfd version 2.39 and newer new warnings are added. Skip those since we
+# are not loaded by a elf loader.
+TF_LDFLAGS             +=      --no-warn-rwx-segments
 TF_LDFLAGS             +=      --fatal-warnings -O1
 TF_LDFLAGS             +=      --gc-sections
 # ld.lld doesn't recognize the errata flags,

```

But a bump to a newer ATF version is probably the more sustainable solution.

thochstein, 2023-05-26:
Hi @dwretman. I don't understand why you would see a problem here. It is working fine for me, and I do see a (different) fix for this in the built source: https://github.com/nxp-imx/imx-atf/commit/de873017ae662ad4d99d5c72bb65ea1bc31eeb83

dwretman, 2023-05-26:
Ah, looking more carefully I found a bbappend overriding this and pointing to an (old) upstream ATF commit.

My bad. Sorry for the noise!

closed 2023-05-26 by dwretman