Freescale/meta-freescale #1561: xwayland: glamor support broken on iMX8

embdur, 2023-06-05:
xwayland glamor support appears to be broken on iMX8:
```
Disabling glamor and dri3 support, XWAYLAND_NO_GLAMOR is set
Failed to initialize glamor, falling back to sw
```

I found this string in libEGL(imx-gpu-viv):
```
# fgrep XWAYLAND_NO_GLAMOR /usr/lib/aarch64-linux-gnu/libEGL.so.1.5.0
grep: /usr/lib/aarch64-linux-gnu/libEGL.so.1.5.0: binary file matches
```
If we change this string to something else e.g. by using sed, Xwayland starts fine with glamor support.

There is no reference to Glamor on https://www.nxp.com/docs/en/user-guide/IMX_GRAPHICS_USERS_GUIDE.pdf or https://github.com/nxp-imx/imx-manifest/blob/imx-linux-kirkstone/GraphicsChangeLogv6. Does NXP even support glamor?

thochstein, 2023-06-06:
Sorry for the trouble. I'm told that glamor was removed due to a conflict with Weston that caused an xterm hang.

We will continue to look at the issue to see if we can restore glamor in a future release.

otavio, 2023-10-12:
As NXP binaries do not support this, there isn't much we can do. Therefore, I will be closing this issue.

closed 2023-10-12 by otavio