Freescale/meta-freescale #1572: imx-atf: broken in imx8mq-evk machine with gcc-13

quaresmajose, 2023-06-13:
The `imx-atf` is broken on imx8mq-evk machine with the new version of gcc 13:
```
|   LD      /srv/oe/build/tmp-lmp/work/imx8mq_evk-lmp-linux/imx-atf/2.6+gitAUTOINC+616a4588f3-r0/git/build-optee/imx8mq/release/bl31/bl31.elf
| aarch64-lmp-linux-ld.bfd: /srv/oe/build/tmp-lmp/work/imx8mq_evk-lmp-linux/imx-atf/2.6+gitAUTOINC+616a4588f3-r0/git/build-optee/imx8mq/release/bl31/bl31.elf section `coherent_ram' will not fit in region `RAM'
| aarch64-lmp-linux-ld.bfd: BL31 image has exceeded its limit.
| aarch64-lmp-linux-ld.bfd: region `RAM' overflowed by 4096 bytes
| aarch64-lmp-linux-ld.bfd: warning: /srv/oe/build/tmp-lmp/work/imx8mq_evk-lmp-linux/imx-atf/2.6+gitAUTOINC+616a4588f3-r0/git/build-optee/imx8mq/release/bl31/bl31.elf has a LOAD segment with RWX permissions
```

reverting back to lf-5.15.71-2.2.0 https://github.com/Freescale/meta-freescale/commit/37264d7854779f15074d5d223f6d8b1acc7cc39d didn't solve the issue.

zandrey, 2023-06-13:
@quaresmajose 

Last time I looked into upstream _TF-A_, commit [`e3c07d2f5a08 ("docs(imx8m): update build support for imx8mq")`](http://git.trustedfirmware.org/TF-A/trusted-firmware-a.git/commit/?id=e3c07d2f5a082d8bb1684ca026d1789a77b3c870) states that support of this platform has been abandoned exactly for the reason of limited _OCRAM_.

I was not able to build the upstream _TF-A_ for `imx8mq` with having same build errors that you observe now for the fork.

Upstream commit says that NXP fork shall still have support for that platform, but with this latest error - it might need a bit more investigations from their side.

-- andrey

quaresmajose, 2023-06-14:
Thanks for the clarifications.
Unfortunately is not possible to open issues on the official [imx-atf](https://github.com/nxp-imx/imx-atf) fork because the tab is disabled.

JackyBai, 2023-06-15:
@quaresmajose can you help provide your build env?

quaresmajose, 2023-06-15:
@JackyBai this issue can be reproduced using the upstream `poky` (after [gcc-13](https://git.yoctoproject.org/poky/commit/?id=17085d769f40537937bf4ec20b4651fc8fc74e2f)) and `meta-freescale` layers master branches with the default configuration so is just `MACHINE=imx8mq-evk bitbake imx-atf`
```
git clone https://git.yoctoproject.org/poky && cd poky
git clone https://github.com/Freescale/meta-freescale
. oe-init-build-env
bitbake-layers add-layer ../meta-freescale
cat << EOF >>conf/local.conf
BB_HASHSERVE_UPSTREAM = "hashserv.yocto.io:8687"
SSTATE_MIRRORS = "file://.* http://sstate.yoctoproject.org/all/PATH;downloadfilename=PATH"
EOF
MACHINE=imx8mq-evk bitbake imx-atf
```

quaresmajose, 2023-06-15:
My results is different now and it build without issue so it can be something in my own distro
```
Loading cache: 100% |                                                                                                                                                                         | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################################| Time: 0:00:13
Parsing of 1050 .bb files complete (0 cached, 1050 parsed). 1973 targets, 175 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.4.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "arch"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8mq-evk"
DISTRO               = "poky"
DISTRO_VERSION       = "4.2"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:3c73bbf9663d9860b729d137b8d32c9b3ea7206f"
meta-freescale       = "master:61c96bc314a271f4f149e0b18616f04a34af2a6c"

NOTE: Fetching uninative binary shim http://downloads.yoctoproject.org/releases/uninative/4.0/x86_64-nativesdk-libc-4.0.tar.xz;sha256sum=fd75b2a1a67a10f6b7d65afb7d0f3e71a63b0038e428f34dfe420bb37716558a (will check PREMIRRORS first)
Initialising tasks: 100% |#####################################################################################################################################################################| Time: 0:00:30
Checking sstate mirror object availability: 100% |#############################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 176 Local 0 Mirrors 118 Missed 58 Current 0 (67% match, 0% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 695 tasks of which 665 didn't need to be rerun and all succeeded.
```
close for now and sorry for the noise

angolini, 2023-06-15:
With `DISTRO ?= 'fslc-xwayland'`it still fails.

angolini, 2023-06-15:
I'm testing with `DISTRO ?= 'fsl-xwayland'` now

angolini, 2023-06-15:
```
Build Configuration:
BB_VERSION           = "2.4.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fsl-linux"
MACHINE              = "imx8mq-evk"
DISTRO               = "fsl-xwayland"
DISTRO_VERSION       = "4.1-snapshot-20230615"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:3c73bbf9663d9860b729d137b8d32c9b3ea7206f"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:45a8bb2620b6a035ae9f42434ab1fde45e0f30cb"
meta-freescale       = "master:61c96bc314a271f4f149e0b18616f04a34af2a6c"
meta-freescale-3rdparty = "HEAD:da0c1b9228db2cf4eb3fe957b8880beb3a301361"
meta-freescale-distro = "HEAD:b09c18a3f649f0cf1b40432bb39e76ba16fe94e2"
```
fails

quaresmajose, 2023-07-03:
After more investigation found that this issue is only reproducible when using the `IMX_DEFAULT_BSP = "imx"` and it can  reproduced on the meta-freescale-distro on the `fsl-xxxxx` distro flavor, `poky` and the the `fslc-xxxxx` _comunity_ is not affected because they use the `IMX_DEFAULT_BSP = "mainline"`.

Step to reproduce:
```
git clone https://git.yoctoproject.org/poky && cd poky
git clone https://github.com/Freescale/meta-freescale
git clone https://github.com/Freescale/meta-freescale-distro
cat << '__EOF__' >helper.sh
#!/bin/bash

set -e

_TOPDIR=$PWD

_build(){
export MACHINE=imx8mq-evk
export DISTRO=$1
cd $_TOPDIR
source oe-init-build-env build-$DISTRO
bitbake-layers add-layer ../meta-freescale
bitbake-layers add-layer ../meta-freescale-distro
cat << EOF >>conf/local.conf
DL_DIR = "${_TOPDIR}/downloads"
SSTATE_DIR = "${_TOPDIR}/sstate-cache"
BB_HASHSERVE_UPSTREAM = "hashserv.yocto.io:8687"
SSTATE_MIRRORS = "file://.* http://sstate.yoctoproject.org/all/PATH;downloadfilename=PATH"
# mask meta-openembedded/meta-oe/recipes-graphics/glm
BBMASK = "/meta-freescale-distro/recipes-graphics/glm/glm_%.bbappend"
IMX_DEFAULT_BSP:fsl = "mainline"
EOF
bitbake imx-atf
}

_build poky
_build fslc-framebuffer
_build fslc-wayland
_build fslc-x11
_build fslc-xwayland
_build fsl-framebuffer
_build fsl-wayland
_build fsl-xwayland
__EOF__
bash helper.sh
```
Removing the config line `IMX_DEFAULT_BSP:fsl = "mainline"` will show the issue.

quaresmajose, 2023-07-04:
The root cause of the issue is because IMX bsp build optee and mainline not

- `IMX_DEFAULT_BSP=imx` => `BUILD_OPTEE=true`

- `IMX_DEFAULT_BSP=mainline` => `BUILD_OPTEE=false`

So the error comes in the `do_compile` task of `recipes-bsp/imx-atf/imx-atf_2.6.bb`:
```
do_compile() {
    # Clear LDFLAGS to avoid the option -Wl recognize issue
    oe_runmake bl31
    if ${BUILD_OPTEE}; then
        oe_runmake clean BUILD_BASE=build-optee
        oe_runmake BUILD_BASE=build-optee SPD=opteed bl31
    fi
}
```

When building the optee `bl31.elf`  in `.../git/build-optee/...`:
```
| aarch64-lmp-linux-ld.bfd: /srv/oe/build/tmp-lmp/work/imx8mq_evk-lmp-linux/imx-atf/2.6+gitAUTOINC+616a4588f3-r0/git/build-optee/imx8mq/release/bl31/bl31.elf section `coherent_ram' will not fit in region `RAM'
```

dv1, 2023-07-04:
@quaresmajose I tried that workaround, but I get over UART the `spl_load_fit_image: Skip load 'atf-1': image size is 0!` line, and then, nothing else happens. The system won't start.

quaresmajose, 2023-07-04:
> @quaresmajose I tried that workaround, but I get over UART the `spl_load_fit_image: Skip load 'atf-1': image size is 0!` line, and then, nothing else happens. The system won't start.

With `BUILD_OPTEE=true` there is not enough space to accommodate the binaries on this `imx8mq-evk` machine.
In my opinion the fix has to be in the source code and not in the meta-layer. However I have no ideia how to fix this issue and my steps were more way to describe who can we replicate the issue. I also use a workaround but just for build I won't even try to use it on a board.

I take the opportunity to say that I also tested using the version [2.8](https://github.com/nxp-imx/imx-atf/tree/lf_v2.8) but it still fail to build with the same error.

JackyBai, 2023-07-19:
> can you unset the CFLAGS before bitbake imx-atf to have a try? for TF-A build, all the CFLAGS set by yocto env should be unset.

angolini, 2023-07-19:
> > can you unset the CFLAGS before bitbake imx-atf to have a try? for TF-A build, all the CFLAGS set by yocto env should be unset.

as far as I know, they are unset already

thochstein, 2023-07-19:
Is this still reproducible? I saw the failure back on June 27 with the following steps:

```
$ mkdir fsl-community-bsp
$ cd fsl-community-bsp
fsl-community-bsp$ repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b master
fsl-community-bsp$ repo sync
fsl-community-bsp$ DISTRO=fsl-xwayland MACHINE=imx8mq-mek . setup-environment build
fsl-community-bsp/build$ bitbake imx-atf
```

It works fine for me today. I also tried the [steps from @quaresmajose](https://github.com/Freescale/meta-freescale/issues/1572#issuecomment-1619180803). That also works fine.

angolini, 2023-07-25:
I have tested it also on my side, and I could NOT reproduce the error anymore. I don't know to explain why.

quaresmajose, 2023-07-31:
All good again on master branch
Not possible to replicate on current master with the last affects `fsl-xxxxx` distro flavor. 
```
git clone https://git.yoctoproject.org/poky && cd poky
git clone https://github.com/Freescale/meta-freescale
git clone https://github.com/Freescale/meta-freescale-distro
cat << '__EOF__' >helper.sh
#!/bin/bash

set -e

_TOPDIR=$PWD

_build(){
export MACHINE=imx8mq-evk
export DISTRO=$1
cd $_TOPDIR
source oe-init-build-env build-$DISTRO
bitbake-layers add-layer ../meta-freescale
bitbake-layers add-layer ../meta-freescale-distro
cat << EOF >>conf/local.conf
DL_DIR = "${_TOPDIR}/downloads"
SSTATE_DIR = "${_TOPDIR}/sstate-cache"
BB_HASHSERVE_UPSTREAM = "hashserv.yocto.io:8687"
SSTATE_MIRRORS = "file://.* http://sstate.yoctoproject.org/all/PATH;downloadfilename=PATH"
# mask meta-openembedded/meta-oe/recipes-graphics/glm
BBMASK = "/meta-freescale-distro/recipes-graphics/glm/glm_%.bbappend"
EOF
bitbake imx-atf
}

_build fsl-framebuffer
_build fsl-wayland
_build fsl-xwayland
__EOF__
bash helper.sh
```

closed 2023-07-31 by quaresmajose