Freescale/meta-freescale #1588: Upgrade to NXP BSP release L6.1.22-2.0.0

otavio, 2023-06-27:
The immediate items to work on are:

- [ ] Upgrade Linux kernel and U-Boot (i.MX variants)
- [ ] Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants) 
- [ ] Upgrade OP-TEE and ATF (i.MX variants)
- [ ] Upgrade Multimedia Packages
- [ ] Upgrade Graphics Packages

The Release Notes are [here](https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf) for checking.

dv1, 2023-06-30:
I'm looking at libimxdmabuffer, libimxvpuapi, and gstreamer-imx upgrades, along with improvements to the way `PACKAGECONFIG_GL` is set in gstreamer1.0-plugins-base. And I think these changes will be possible to backport to kirkstone and dunfell.

MrCry0, 2023-07-04:
@otavio would you mind if I take the first task? Upgrade Linux kernel and U-Boot (i.MX variants)

otavio, 2023-07-05:
@MrCry0, I added you to the description.

dv1, 2023-08-01:
libimxdmabuffer, libimxvpuapi2, gstreamer-imx upgraded. PRs:

https://github.com/Freescale/meta-freescale/pull/1630
https://github.com/Freescale/meta-freescale/pull/1631

closed 2023-08-21 by otavio