Freescale/meta-freescale #1593: Updated to latest kirkstone has broken wic/etx4 image generation

JPSGoncalves, 2023-06-30:
I update my meta-freescale layer to latest kirskstone branch of a project that is based on Yocto kirkstone release. After this update i can not generate wic images or even ext4 rootfs images. I needed to updated because of libdrm-imx recipe for new fetch path.  All the build steps go without errors, only the wic/ext4 image generation gives a segmentation fault. I could not find what is causing this, so i am asking for help. 

The error messages is this 

`ERROR: _exec_cmd: export PATH=/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/recipe-sysroot-native/sbin:/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/recipe-sysroot-native/usr/sbin:/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/recipe-sysroot-native/usr/bin:/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/recipe-sysroot-native/bin:/home/user/yocto/workspace/build-xcomm/tmp/hosttools:$PATH;export PSEUDO_PREFIX=/home/user/yocto/workspace/build-xcomm/tmp/sysroots-components/x86_64/pseudo-native/usr;export PSEUDO_LOCALSTATEDIR=/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/rootfs/../pseudo;export PSEUDO_PASSWD=/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/rootfs;export PSEUDO_NOSYMLINKEXP=1;export PSEUDO_IGNORE_PATHS=/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/tmp-wic/rootfs_root.2.ext4,/usr/,/etc/,/lib,/dev/,/run/,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/temp,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/recipe-sysroot,/home/user/yocto/workspace/build-xcomm/sstate-cache,/home/user/yocto/workspace/build-xcomm/tmp/stamps,/home/user/yocto/workspace/build-xcomm/tmp/sstate-control,/home/user/yocto/workspace/build-xcomm/tmp/buildstats,/home/user/yocto/workspace/build-xcomm/tmp/sysroots-components,/home/user/yocto/workspace/build-xcomm/tmp/pkgdata,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/deploy-,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/sstate-build-package_,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/sstate-install-package_,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/pkgdata-sysroot,/home/user/yocto/workspace/build-xcomm/tmp/deploy,/home/user/yocto/workspace/build-xcomm/cache,/home/user/yocto/workspace/poky/scripts,/home/user/yocto/workspace/poky/meta/files/common-licenses,/home/user/yocto/workspace/meta-openembedded/meta-oe/licenses,/home/user/yocto/workspace/meta-openembedded/meta-multimedia/licenses,/home/user/yocto/workspace/meta-openembedded/meta-python/licenses,/home/user/yocto/workspace/meta-freescale/custom-licenses,/home/user/yocto/workspace/meta-python2/licenses,/home/user/yocto/workspace/poky/meta/COPYING,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/x86_64-deploy-simple-image-ai-dev-populate-sdk,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/oe-sdk-repo,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/sstate-build-populate_sdk,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/build-wic,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/intercept_scripts,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/oe-rootfs-repo,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/sstate-build-image_complete,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/simple-image-ai-dev-1.0,/home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/simple-image-ai-dev-1.0;/home/user/yocto/workspace/build-xcomm/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo mkfs.ext4 -F -i 8192 /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/tmp-wic/rootfs_root.2.ext4 -L root -U 24dea335-a0a1-40a6-a82d-b714fd0ff57e -d /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/rootfs returned '139' instead of 0
| output: /bin/sh: line 1: 287904 Segmentation fault      (core dumped) /home/user/yocto/workspace/build-xcomm/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo mkfs.ext4 -F -i 8192 /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/tmp-wic/rootfs_root.2.ext4 -L root -U 24dea335-a0a1-40a6-a82d-b714fd0ff57e -d /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/rootfs
| WARNING: /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/temp/run.do_image_wic.287884:161 exit 1 from 'BUILDDIR="/home/user/yocto/workspace/build-xcomm" PSEUDO_UNLOAD=1 wic create "$wks" --vars "/home/user/yocto/workspace/build-xcomm/tmp/sysroots/xcomm/imgdata/" -e "simple-image-ai-dev" -o "$build_wic/" -w "$tmp_wic" --no-fstab-update'
| WARNING: Backtrace (BB generated script):
| 	#1: do_image_wic, /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/temp/run.do_image_wic.287884, line 161
| 	#2: main, /home/user/yocto/workspace/build-xcomm/tmp/work/xcomm-poky-linux/simple-image-ai-dev/1.0-r0/temp/run.do_image_wic.287884, line 179
ERROR: Task (/home/user/yocto/workspace/meta-my-layer/recipes-core/images/simple-image-ai-dev.bb:do_image_wic) failed with exit code '1'
`
One thing that i noticed is that the tools mkfs.ext4 from the sysroot-native of the recipe when executed on gives a segmentation fault. 

Can you help me figure out the possible problems that are causing this?

Thanks!

paletteguy, 2023-07-03:
Have the same error after update.

otavio, 2023-07-03:
Have you tried running a bisect to determine which commit is causing the error?

paletteguy, 2023-07-04:
My problem is only build server using ubuntu 18.04.  On my own computer with kubuntu 22.04 I do not have the problem. Strange.. since tools like mkfs.ext4 should be compiled natively by bitbake and when used. It does run "pseudo mkfs.ext4" it could be the pseudo environment causing this.

paletteguy, 2023-07-04:
Looks like the problem comes from poky. After reverting poky to kirkstone-4.0.10 I  not getting seg-fault when doing wic.

paletteguy, 2023-07-04:
Using git bisect I found hash failing in poky, 263d5df53922ddf6d00859b9a88d1f9fe1f851bd

It looks like the kernel 5.15.110 is causing the issue for qemu.

paletteguy, 2023-07-04:
@shr-project might have provided the work around for the problem since I did a new bisect and failed elsewhere. 

I'm testing this out now.. https://patchwork.yoctoproject.org/project/oe-core/list/?series=14023 it will take awhile to compile I'll get back with more info.

shr-project, 2023-07-04:
Yes, mkfs.ext4 segfault is caused by uninative upgrade which includes new
patchelf-uninative which breaks the binaries.

paletteguy, 2023-07-04:
@shr-project When will it be available in Kirkstone ?

shr-project, 2023-07-04:
Probably not anytime soon as it was rejected from master, but it's easy to
backport by overlaying uninative.bbclass, that's what I did in projects I
care about (and which still use unsupported 18.04 where it's fatal issue).

paletteguy, 2023-07-04:
@shr-project Thx for your support :)

paletteguy, 2023-07-05:
This issue can be closed. @otavio

closed 2023-07-05 by otavio