Freescale/meta-freescale #1626: Support Yocto mickledore?

Livius90, 2023-07-27:
NXP already published mickledore meta-imx BSP layer, when will be mickledore supported in meta-freescale on GitHub?
https://github.com/nxp-imx/meta-imx/tree/mickledore-6.1.22-2.0.0

otavio, 2023-07-27:
We already support it on the master branch and it is compatible with multiple releases.

Livius90, 2023-07-29:
Not this is the official and professional way to support a Yocto release, you shall make a mickledore branch to support it for mickledore. When it will be done? Master barnch is master branch for developing example not for Yocto releases it is totally untrustworthy to use.

otavio, 2023-07-31:
Please check https://github.com/Freescale/meta-freescale/discussions/1329.

otavio, 2023-07-31:
Next branch intended is for next LTS release.

Livius90, 2023-08-25:
answered

closed 2023-08-25 by Livius90