Freescale/meta-freescale #1629: G2D does not work with current git master

dv1, 2023-07-31:
I tried to test G2D code on the imx6 (machine is imx6qdlsabresd). meta-freescale is at commit 992973ab4bdbabd21a2c1f2cea86d85e60a79c5e . G2D does not work due to a userspace<->kernelspace mismatch:

```
[     1] HAL user version: 6.4.11.608260
[     2] HAL kernel version: 6.4.11.658245
```

Is this known? @otavio  @thochstein is a new version of imx-gpu-g2d being prepared?

otavio, 2023-07-31:
It is part of #1588; it seems it has not been sent yet.

dv1, 2023-08-01:
FYI, for anyone who needs to test-run git master, reverting ce7a697d256f32a73135b22bf744228ef6df3315 worked for me as a stopgap solution.

thochstein, 2023-08-01:
I thought, @otavio, that you mentioned someone else was working on it? Shall I work on it next?

closed 2023-08-01 by otavio