Freescale/meta-freescale #1653: Handle multiple values for IMXBOOT_TARGETS

angolini, 2023-09-26:
Thanks @bmx666 and @angolini, this sounds very useful.

Can we extend `UBOOT_CONFIG` directly?

```
# MACHINE M1 and M2
KERNEL_DEVICETREE = "\
    freescale/imx8mp_m1.dtb \
    freescale/imx8mp_m2.dtb \
"
IMXBOOT_TARGETS = "flash_evk"
UBOOT_CONFIG = "sd-m1 sd-m2"

UBOOT_CONFIG[sd-m1] = "imx8mp_m1_defconfig,sdcard,imx8mp_m1.dtb"
UBOOT_CONFIG[sd-m2] = "imx8mp_m2_defconfig,sdcard,imx8mp_m2.dtb"

WKS_FILE = "${IMAGE_BASENAME}.wks.in"
```

Can we add `IMXBOOT_TARGETS` to `UBOOT_CONFIG`?

We should have an actual implementation. The EVK machines seem a good fit. How about reworking the `include/imx*-evk.inc` design? For an example, if we choose `imx8mp-unified-evk.conf` for the unified machine name, then `imx8mp-lpddr4-evk.conf` might include the following important lines:

```
require imx8mp-unified-evk.conf
UBOOT_CONFIG = "sd-lpddr4"
```

_Originally posted by @thochstein in https://github.com/Freescale/meta-freescale/issues/1642#issuecomment-1697936744_

thochstein, 2023-09-26:
> UBOOT_CONFIG[sd-m1] = "imx8mp_m1_defconfig,sdcard,imx8mp_m1.dtb"
> UBOOT_CONFIG[sd-m2] = "imx8mp_m2_defconfig,sdcard,imx8mp_m2.dtb"

This is wrong, there is already a third field which is just now being documented: https://lists.yoctoproject.org/g/poky/message/13167

angolini, 2023-09-26:
I'm not sure which exact comment to pin here, that's why I pinnet this one. However, the main point of this issue is to discuss how to handle multiple IMXBOOT_TARGETS and how to related the IMXBOOT_TARGET to the machine

thochstein, 2023-09-26:
Okay. I did just fix my original comment.