Freescale/meta-freescale #1654: build failed: wayland-scanner found: NO

DominikPoggel, 2023-09-28:
I tried building an image with the L6.1.36-2.1.0 BSP, but it fails with wayland-scanner:

| Found pkg-config: /home/pog/OSM/imx-yocto-bsp/build_iesy-wayland/tmp/work/cortexa53-crypto-iesy-linux/wayland/1.21.0-r0/recipe-sysroot-native/usr/bin/pkg-config-native (0.29.2)
| Dependency wayland-scanner found: NO found 1.22.0 but need: '1.21.0'
| Found CMake: /home/pog/OSM/imx-yocto-bsp/build_iesy-wayland/tmp/work/cortexa53-crypto-iesy-linux/wayland/1.21.0-r0/recipe-sysroot-native/usr/bin/cmake (3.25.2)
| Build-time dependency wayland-scanner found: NO (tried cmake)
|
| ../wayland-1.21.0/src/meson.build:81:1: ERROR: Dependency lookup for wayland-scanner with method 'pkgconfig' failed: Invalid version, need 'wayland-scanner' ['1.21.0'] found '1.22.0'.

quaresmajose, 2023-09-28:
Why are you building the wayland-1.21.0? According to with https://github.com/Freescale/meta-freescale/pull/1589 the layer would picking up 1.22.0.

otavio, 2023-10-12:
This appears to be using NXP's release, so it is recommended to report any errors in their https://github.com/nxp-imx/meta-imx repository. This ticket will be closed.

closed 2023-10-12 by otavio

Zervandez, 2023-12-17:
@DominikPoggel how did you solve this issue? I've encountered the same...