Freescale/meta-freescale #1655: Upgrade BSP to LF6.1.36_2.1.0

otavio, 2023-09-28:
The immediate items to work on are:

- [ ] Upgrade Linux kernel and U-Boot (i.MX variants) 
- [ ] Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants) 
- [ ] Upgrade OP-TEE
- [ ]  ATF (i.MX variants) 
- [ ] Upgrade Multimedia Packages
- [ ] Upgrade Graphics Packages
- [ ] Jailhouse
- [ ] mcore-demos

The Release Notes are [here](https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf) for checking.

MrCry0, 2023-10-08:
@otavio we're preparing the next release so that I can take action on these tasks next week or so.

angolini, 2023-10-11:
I'm only organizing the remaining work on kernel:

### linux-fslc-imx_6.1.bb 
From `KBRANCH = "6.1-2.0.x-imx"` to  `KBRANCH = "6.1-2.1.x-imx`
* https://github.com/Freescale/linux-fslc/pull/643

### linux-fslc_6.1.bb
Keep the branch `KBRANCH = "6.1.x+fslc"` update the kernel stable version from `v6.1.38` to `v6.1.57`
* https://github.com/Freescale/linux-fslc/pull/644

### linux-imx_6.1.bb
Keep the branch `SRCBRANCH = "lf-6.1.y"`. But change the `LOCALVERSION` from `-6.1.22-2.0.0` to `-6.1.36-2.1.0`

zandrey, 2023-10-30:
@otavio You can put me to the Task 3 _Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants)_ in the list.

I'm planning to push Kernel `linux-6.6.y` and U-Boot `v2023.10` into forks and layer.

angolini, 2023-11-07:
Is #1679 enough to close the graphic task?

angolini, 2023-12-11:
I just updated the list of packages to include firmware-sentinel and imx-mkimage after the build error for imx93

ricardosalveti, 2023-12-18:
Headers was missing, required by imx-pxp-g2d: https://github.com/Freescale/meta-freescale/pull/1710

PR #1688 (2023-11-08): Multimedia Update for NXP release 6.1.36-2.1.0
Fixes #1655
closed 2024-01-25 by angolini