Freescale/meta-freescale #1665: opencv_4.6.0.imx not building with newest meta-openembedded protobuf version upgrade

lms-ts, 2023-10-12:
With this commit of meta-openembedded
https://github.com/openembedded/meta-openembedded/commit/32901e95b88f2867cc2aec28d0b2fee19957ca3c

protobuf is upgraded from version 3.x to 4.x. This breaks the build of opencv_4.6.0.imx with errors like

```bash
| /home/tschuster/avnet/build/tmp/work/cortexa53-crypto-mx8mp-lms-linux/opencv/4.6.0.imx/recipe-sysroot/usr/include/google/protobuf/map.h:778:7: error: 'Tree' was not declared in this scope; did you mean 'free'?
|   778 |       Tree* tree = internal::TableEntryToTree<Tree>(table_[b]);
|       |       ^~~~
|       |       free
| /home/tschuster/avnet/build/tmp/work/cortexa53-crypto-mx8mp-lms-linux/opencv/4.6.0.imx/recipe-sysroot/usr/include/google/protobuf/map.h:778:13: error: 'tree' was not declared in this scope; did you mean 'free'?
|   778 |       Tree* tree = internal::TableEntryToTree<Tree>(table_[b]);
|       |             ^~~~
|       |             free
| /home/tschuster/avnet/build/tmp/work/cortexa53-crypto-mx8mp-lms-linux/opencv/4.6.0.imx/recipe-sysroot/usr/include/google/protobuf/map.h:778:52: error: no matching function for call to 'TableEntryToTree<Tree>(google::protobuf::internal::TableEntryPtr&)'
|   778 |       Tree* tree = internal::TableEntryToTree<Tree>(table_[b]);
|       |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
```
Building against the previous commit of meta-openembedded (045e7df0f47b0abf4648748fc679e8daac648490) works just fine.

quaresmajose, 2023-10-12:
Some opencv 4.8 fixes lands on master can you please try https://github.com/openembedded/meta-openembedded/commit/225ce6a14a8110ab6b573b4dc9f5297a03d17e0f to see if it solves also this issue.

lms-ts, 2023-10-12:
Changing to that meta-openembedded commit does not solve the issue, as the opencv_4.6.0.imx.bb recipe does not contain that patch. 
But sounds like integrating that patch into the opencv_4.6.0.imx.bb recipe will solve the issue.

lms-ts, 2023-10-12:
I integrated the patch into the opencv_4.6.0.imx.bb recipe and it solves the issue with dnn enabled.