Freescale/meta-freescale #1671: The Yocto mirror is out of sync

quaresmajose, 2023-10-20:
The Yocto mirror is out of sync since 2023-10-09 

https://git.yoctoproject.org/meta-freescale

All of this commits are missing in the mirror
```
647dddd6 (up/master) Merge pull request #1668 from quaresmajose/firmware-nxp-wifi
128e9726 firmware-nxp-wifi: make it compatible with imx-generic-bsp
d06f10d8 Merge pull request #1667 from angolini/bump-linux-imx-2.1.0
d0eed117 Merge pull request #1670 from angolini/bump-linux-fslc-imx-2.1.0
37c9deb2 linux-fslc-imx: Update to lf-6.1.36-2.1.0
1fb759b2 Merge pull request #1669 from angolini/bump-linux-fslc
e1702615 linux-fslc: update to v6.1.57
3ba46481 imx93-9x9-lpddr4-qsb: Include dts for lf-6.1.36_2.1.0
7ebfbdc9 imx8qxp-mek: Include dts for lf-6.1.36_2.1.0
b35c2cac imx8mp-lpddr4-evk: Include dts for lf-6.1.36_2.1.0
0eac105a linux-imx: Update lf-6.1.22_2.0.0 to lf-6.1.36_2.1.0
7327e03c ls1021atwr: drop as it is unbuildable for too long
a8d8dc46 ls2080ardb: drop as it is unbuildable for too long
769e3c76 Merge pull request #1666 from lms-ts/master
e0bdbc61 opencv: Integrate https://github.com/openembedded/meta-openembedded/commit/225ce6a14a8110ab6b573b4dc9f5297a03d17e0f to fix build protobuf v22 and dnn enabled
d441dcbe Merge pull request #1662 from hiagofranco/update_uuu_src_link
a48f5ee3 uuu: Update nxp repository links
```

quaresmajose, 2023-10-20:
The yocto mirror moves recently to another server and in this step they change all the url so soemthing needs to changed somewhere 
```
-https://git.yoctoproject.org/git/meta-freescale
+https://git.yoctoproject.org/meta-freescale
```

otavio, 2023-10-20:
I notified them, thanks for letting us know.

otavio, 2023-10-20:
It seems fine now. Closing.

closed 2023-10-20 by otavio