Freescale/meta-freescale #1686: [gpu] weston fails to start using 5.15 kernel

otavio, 2023-11-03:
@thochstein this seems to regress with 5.15 kernel. I am using varscite SOMs which are still on 5.15.60 kernel. weston refuses to start 

```
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.350] weston 11.0.2
Nov 03 05:32:21 imx8qm-var-som weston[1691]:                https://wayland.freedesktop.org
Nov 03 05:32:21 imx8qm-var-som weston[1691]:                Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
Nov 03 05:32:21 imx8qm-var-som weston[1691]:                Build: 10.0.0-661-ga3c5e63a+
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.350] Command line: /usr/bin/weston --modules=systemd-notify.so
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.350] OS: Linux, 5.15.60-imx8qm-imx8qm+g740e6c7a7b09, #1 SMP PREEMPT Fri Jul 14 11:33:35
 UTC 2023, aarch64
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.350] Flight recorder: enabled
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.350] Using config file '/etc/xdg/weston/weston.ini'
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.351] Output repaint window is 16 ms maximum.
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.351] Loading module '/usr/lib/libweston-11/drm-backend.so'
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.357] initializing drm backend
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.357] Trying logind launcher...
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.377] logind: session control granted
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.394] using /dev/dri/card0
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.394] DRM: supports atomic modesetting
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.394] DRM: supports GBM modifiers
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.394] DRM: supports picture aspect ratio
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.394] Loading module '/usr/lib/libweston-11/gl-renderer.so'
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [     1] HAL user version: 6.4.11.684571
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [     2] HAL kernel version: 6.4.11.658245
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.403] failed to initialize display
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.403] EGL error state: EGL_NOT_INITIALIZED (0x3001)
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.403] failed to initialize egl
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [05:32:21.403] fatal: failed to create compositor backend
Nov 03 05:32:21 imx8qm-var-som weston[1691]: Internal warning: debug scope 'drm-backend' has not been destroyed.
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [     3] HAL user version: 6.4.11.684571
Nov 03 05:32:21 imx8qm-var-som weston[1691]: [     4] HAL kernel version: 6.4.11.658245
Nov 03 05:32:21 imx8qm-var-som systemd[1]: weston.service: Main process exited, code=exited, status=1/FAILURE
Nov 03 05:32:21 imx8qm-var-som systemd[1]: weston.service: Failed with result 'exit-code'.
Nov 03 05:32:21 imx8qm-var-som systemd[1]: Failed to start Weston, a Wayland compositor, as a system service.
```

Any ideas ?

_Originally posted by @kraj in https://github.com/Freescale/meta-freescale/issues/1679#issuecomment-1791931634_

otavio, 2023-11-03:
Yes, please upgrade https://github.com/Freescale/kernel-module-imx-gpu-viv/blob/master/kernel-module-imx-gpu-viv-src/hal/kernel/inc/gc_hal_version.h#L67 and see if it helps.

thochstein, 2023-11-03:
Oh, sorry, I did forget to upgrade`kernel-module-imx-gpu-viv`, will do it now.

kraj, 2023-11-03:
FWIW https://github.com/Freescale/meta-freescale/pull/1687 fixed this problem. Thanks @thochstein

closed 2023-11-07 by otavio