Freescale/meta-freescale #1715: Upgrade BSP to LF6.1.55_2.2.0

angolini, 2023-12-29:
The immediate items to work on are:

- [ ] Upgrade Linux kernel and headers (i.MX variants)
- [ ] U-Boot (i.MX variants)
- [ ] Upgrade Linux kernel and U-Boot (FSLC forks i.MX variants)
- [ ] Upgrade OP-TEE
- [ ] ATF (i.MX variants)  
- [ ] Upgrade Multimedia Packages
- [ ] Upgrade Graphics Packages 
- [ ] Jailhouse 
- [ ] mcore-demos
- [ ] firmware-sentinel 
- [ ] imx-mkimage

The Release Notes are [here](https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf) for checking.

angolini, 2024-01-12:
Kernel subtasks:

- linux-fslc-imx_6.1.bb
  - [ ] From KBRANCH = "6.1-2.1.x-imx" to KBRANCH = "6.1-2.2.x-imx
  - [ ] kernel headers

- linux-fslc_6.1.bb
  - [ ] Keep the branch KBRANCH = "6.1.x+fslc" update the kernel stable version from v6.1.57 to v6.1.70
  - [ ] kernel headers

- linux-imx_6.1.bb
   - [ ] Keep the branch SRCBRANCH = "lf-6.1.y". But change the LOCALVERSION from -6.1.36-2.1.0 to -6.1.55-2.2.0 - https://github.com/Freescale/meta-freescale/pull/1727
   - [ ] kernel headers - https://github.com/Freescale/meta-freescale/pull/1728

MrCry0, 2024-02-17:
@otavio I prepared the u-boot-fslc 2024.01 but don't have permissions to push it directly to `github.com/Freescale/u-boot-fslc.git`.

Please push the branch `https://github.com/MrCry0/u-boot-fslc/tree/2024.01%2Bfslc` into `Freescale/u-boot-fslc` so that I'll be able to update the u-boot fslc for this issue.

otavio, 2024-02-17:
> @otavio I prepared the u-boot-fslc 2024.01 but don't have permissions to push it directly to `github.com/Freescale/u-boot-fslc.git`.
>
> Please push the branch [...]

Done 👍🏻 thanks!

closed 2024-02-17 by otavio