Freescale/meta-freescale #1725: [Kirkstone] - gstreamer version 1.20.3 and "-D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64" build options

marlene-jacquier, 2024-01-11:
There is an issue in official gstreamer version when building it with "-D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64" options. This has been fixed in gstreamer version 1.20.5 with the following commit: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/commit/94a0d579d7c987dd3fb40eaa579fb6dde8fdc6f4

The symptom is that a gstreamer pipeline is in this case taking 80% of the CPU without the patch, no issue when using 32 bits time.

Is there a plan to integrate at least gstreamer 1.20.5 on the kirkstone branch. to avoid patching it ?

quaresmajose, 2024-01-11:
If the patch you have mention fix the issue you can submit it as backport in https://github.com/Freescale/meta-freescale/blob/88b87bd387c8921b60279d09b22560f9e08fd9c0/recipes-multimedia/gstreamer/gstreamer1.0_1.22.0.imx.bb#L94

something like that will do the job:
```
SRC_URI += "file://my.patch"
```

marlene-jacquier, 2024-01-11:
Actually the issue is already fixed in gstreamer1.0_1.22.0, as I said the patch mentionned is included in gstreamer1.0_1.20.5.
The question is about kirkstone branch, is there a plan to update to at least version 1.20.5 (currently 1.20.3) ?

quaresmajose, 2024-01-11:
Sorry I got confused with the version, the patch should be applied on the https://github.com/Freescale/meta-freescale/blob/47e0b467d8555e38a996b731977d9eed9ab02051/recipes-multimedia/gstreamer/gstreamer1.0_1.20.3.imx.bb#L90 

About the updated on kirk branch, I have no idea about when gstreamer imx fork will be updated.

otavio, 2024-01-11:
I think it would be a good idea to prepare a PR. Could you please prepare it?

thochstein, 2024-01-11:
Hi @marlene-jacquier, can you make a PR with a patch file as @quaresmajose suggests?

marlene-jacquier, 2024-01-19:
PR created: https://github.com/Freescale/meta-freescale/pull/1737

closed 2024-01-29 by otavio