Freescale/meta-freescale #1726: [Kirkstone] - gstreamer-plugins-imx and imx-gst-plugin

marlene-jacquier, 2024-01-11:
In the meta-freescale, there are two differents packages for i.MX gstreamer plugins, what is exactly the difference ?
When using imx-gst-plugin, it seems imxv4l2sink is used in a gstreamer pipeline by default (playbin).
When using gstreamer-plugins-imx, it seem glimagesink is used in a gstreamer pipeline by default (playbin).

On i.MX6 platform, glimagesink is not working properly, maybe because of the video format.
Could you clarify what is the proper configuration to be used on i.MX6 ?

ursfassler, 2025-02-27:
Came across this question while trying to upgrade from Kirkstone to Scarthgap, gstreamer pipeline on i.MX8plus and it didn't work with Scarthgap (glimagesink). I haven't used imx-gst-plugin. Maybe this info helps somebody.

imx-gst-plugin provides the elements:

- aiurdemux:  aiurdemux: IMX Aiur universal demuxer
- aiurdemux: webm: webm
- beepdec: 3ca: ac3
- beepdec: ac3: ac3
- beepdec:  beepdec: IMX Beep universal decoder
- imxcompositor:  imxcompositor_g2d: IMX g2d Video Compositor
- imxvideoconvert:  imxvideoconvert_g2d: IMX g2d Video Converter
- imxvideoconvert:  imxvideoconvert_ocl: IMX ocl Video Converter
- spdifdemux:  spdifdemux: SPDIF demuxer
- vpu:  vpudec: IMX VPU-based video decoder
- vpu:  vpuenc_h264: IMX VPU-based AVC/H264 video encoder
- vpu:  vpuenc_hevc: IMX VPU-based HEVC video encoder

My test-pipeline works now ok with imxvideoconvert_g2d and fbdevsink (good enough to go further, hope that we can use a accelerated sink in the future. (imxvideoconvert_g2d would probably also work with Kirkstone 🤔)

With Kirkstone:
```bash
gst-launch-1.0 filesrc location=big_buck_bunny_720p_10mb.mp4 ! qtdemux ! h264parse ! v4l2h264dec ! videoconvert ! glimagesink
```

With Scarthgap:
```bash
gst-launch-1.0 filesrc location=big_buck_bunny_720p_10mb.mp4 ! qtdemux ! h264parse ! v4l2h264dec ! imxvideoconvert_g2d ! fbdevsink
```

angolini, 2025-02-27:
I tried to copy to discussion but an error happened but i cannot figure it out. Let it be for a while.

I don't know if @dv1 is still working on the first question though