Freescale/meta-freescale #1743: Closed sources binaries and compilation flags

blemouzy, 2024-01-29:
Hello,

I'm working with Yocto Kirkstone on i.MX6 CPU for industrial applications.
BSP and applications must be [Y2038 ready](https://theyear2038problem.com/).
So everything is built with "-D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64" compilation flags.

But some meta-freescale recipes used precompiled binaries / libraries like [imx-gpu-viv](https://www.nxp.com/lgfiles/NMG/MAD/YOCTO/imx-gpu-viv-6.4.3.p4.6-aarch32.bin).

Do you know compilation flags used to build them?
- Are they built with [security flags](https://git.yoctoproject.org/poky/tree/meta/conf/distro/include/security_flags.inc?h=kirkstone)? (enabled by default with Kirkstone)
- Are they built with time64 flags?

thochstein, 2024-01-30:
Hi @blemouzy,

> * Are they built with [security flags](https://git.yoctoproject.org/poky/tree/meta/conf/distro/include/security_flags.inc?h=kirkstone)? (enabled by default with Kirkstone)

The graphics packages mostly do use them, but I don't know about the rest. I suspect they may not. I have made an internal ticket to address this.

> * Are they built with time64 flags?

This issue was not on our radar until Yocto mandated it with Nanbield. We have made some progress on this for our Q1 Nanbield release, but, unfortunately, we do not have a plan to backport any of this to Kirkstone. Otherwise, we hope to make some good progress on both issues for our Q2 release, where we are targeting the next Yocto LTS 5.0 Scarthgap.

blemouzy, 2024-02-05:
Hi @thochstein,

> The graphics packages mostly do use them, but I don't know about the rest. I suspect they may not. I have made an internal ticket to address this.

Thanks!
Please keep us in touch.

> This issue was not on our radar until Yocto mandated it with Nanbield. We have made some progress on this for our Q1 Nanbield release, but, unfortunately, we do not have a plan to backport any of [...]

Do you think that we will able to use these new precompiled libraries with Kirkstone?

closed 2024-04-01 by otavio