Freescale/meta-freescale #1776: plat_prng_add_jitter_entropy fatal error on i.MX93

tprrt, 2024-03-28:
When traces are enabled, at boot, the following error can be observed:

```
D/TC:0 0 core_mmu_xlat_table_alloc:554 xlat tables used 6 / 8
D/TC:? 0 tee_ta_init_pseudo_ta_session:297 Lookup pseudo TA 7011a688-ddde-4053-a5a9-7b3c4ddf13b8
D/TC:? 0 tee_ta_init_pseudo_ta_session:310 Open device.pta
D/TC:? 0 tee_ta_init_pseudo_ta_session:327 device.pta : 7011a688-ddde-4053-a5a9-7b3c4ddf13b8
F/TC:? 0 plat_prng_add_jitter_entropy:72 0x7309
D/TC:? 0 tee_ta_close_session:468 csess 0x2a3e8f40 id 1
D/TC:? 0 tee_ta_close_session:487 Destroy session

```

Moreover `/dev/hwrng` is provided by OP-TEE and doesn't seem to work properly.

tprrt, 2024-04-05:
Hello,
This issue still occurs on lf-6.1.55-2.2.0-0:
```
D/TC:? 0 ldelf_load_ldelf:110 ldelf load address 0xc0007000
D/LD:  ldelf:142 Loading TS ed32d533-99e6-4209-9cc0-2d72cdd998a7
F/TC:? 0 trace_syscall:150 syscall #3 (syscall_get_property)
F/TC:? 0 trace_syscall:150 syscall #5 (syscall_open_ta_session)
D/TC:? 0 ldelf_syscall_open_bin:163 Lookup user TA ELF ed32d533-99e6-4209-9cc0-2d72cdd998a7 (early TA)
D/TC:? 0 ldelf_syscall_open_bin:167 res=0xffff0008
D/TC:? 0 ldelf_syscall_open_bin:163 Lookup user TA ELF ed32d533-99e6-4209-9cc0-2d72cdd998a7 (Secure Storage TA)
I/TC: WARNING (insecure configuration): Failed to get monotonic counter for REE FS, using 0
D/TC:? 0 ldelf_syscall_open_bin:167 res=0xf0100003
D/TC:? 0 ldelf_syscall_open_bin:163 Lookup user TA ELF ed32d533-99e6-4209-9cc0-2d72cdd998a7 (REE)
D/TC:? 0 ldelf_syscall_open_bin:167 res=0xffff0009
E/LD:  init_elf:486 sys_open_ta_bin(ed32d533-99e6-4209-9cc0-2d72cdd998a7)
E/TC:? 0 ldelf_init_with_ldelf:152 ldelf failed with res: 0xffff0009
D/TC:? 0 tee_ta_open_session:694 init session failed 0xffff0009
F/TC:? 0 plat_prng_add_jitter_entropy:72 0xD5
[...]
F/TC:? 0 plat_prng_add_jitter_entropy:72 0x7FEB
```

MrCry0, 2025-04-10:
@tprrt could you please try to reproduce this issue on current freescale codebase with the kernel lf-6.6?