Freescale/meta-freescale #1777: Upgrade BSP to LF6.6.3_1.0.0

MrCry0, 2024-03-29:
The immediate items to work on are:

- [ ]  Upgrade Linux kernel and headers (i.MX variants) 
- [ ]  U-Boot (i.MX variants)
- [ ]  Upgrade Linux kernel (FSLC forks i.MX variants)
- [ ]  Upgrade U-Boot (FSLC forks i.MX variants)
- [ ]  Upgrade OP-TEE
- [ ]  Upgrade ATF (i.MX variants)
- [ ]  Upgrade Multimedia Packages
- [ ]  Upgrade Graphics Packages
- [ ]  Upgrade Jailhouse
- [ ]  Upgrade mcore-demos
- [ ]  Upgrade firmware-imx*
- [ ]  Upgrade imx-mkimage

The Release Notes are [here](https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf) for checking.

angolini, 2024-04-01:
From the SCR it looks like we have some new packages:

- imx-sm.git
- imx-oei.git
- crypto-af-alg.git
- mali-imx-r47.0-b867f39.bin
- imx95-m7-demo-2.15.000.bin
- imx93-m33-demo-2.15.000.bin
- neutron.git
- tflite-neutron-delegate.git

There are other packages that are not yet upstreamed, basically because they are demos or based on cortex M. I don't know how critical they are. I'm only pointing out some new packages to review.

MrCry0, 2024-04-16:
I'm working on u-boot, optee and atf

tprrt, 2024-04-30:
Hello,

It is seems there is a regression on i.MX93:
```
[    1.470460] tee_client_open_session failed, err: ffff000c
[    1.475883] TEE-Crypto: Init failed[0xffffffea].
```

Moreover https://github.com/Freescale/meta-freescale/pull/1809.

Kind regards,
Thomas Perrot

angolini, 2024-05-10:
It looks like this issue is completed and can be closed

closed 2024-05-10 by angolini