Freescale/meta-freescale #1796: No kernel log after migration to Scarthgap

DominikPoggel, 2024-04-23:
I'm currently migrating our BSP on custom hardware from kirkstone to scarthgap, and after porting all kernel patches from linux-fslc-imx 5.15 to 6.1 I'm not seeing any kernel logs. When entering dmesg I also get an empty response. The only output on console is from systemd. 

Did anything change in meta-freescale that might affect the kernel log behaviour?

rafluan, 2024-04-23:
@DominikPoggel Could you give me a commit hash of the meta-freescale that is at the top of your environment? Are you using the nxp or mainline bsp?

DominikPoggel, 2024-04-24:
@rafluan I am on commit 883ac36a4c027049aca32d46690b03a7aa65a6cc.
I'm using the mainline bsp.

@otavio I'm not seeing any commits after 987473 (from 22.04.24), I will try it with that one as it is the newest

DominikPoggel, 2024-04-24:
My Kernel-Log problem seems to be resolved in 6.6-1.0.x-imx, but now mipi_dsi isn't working anymore, in 5.15-2.2.x-imx there were no problems with it.

[    6.542508] imx-drm display-subsystem: bound imx-lcdif-crtc.0 (ops lcdif_crtc_ops)
[    6.550275] imx_sec_dsim_drv 32e10000.mipi_dsi: version number is 0x1060200
[    6.557396] [drm:drm_bridge_attach] *ERROR* failed to attach bridge /soc@0/bus@32c00000/mipi_dsi@32e10000 to encoder DSI-34: -517
[    6.569110] imx_sec_dsim_drv 32e10000.mipi_dsi: Failed to attach bridge: 32e10000.mipi_dsi
[    6.569122] imx_sec_dsim_drv 32e10000.mipi_dsi: failed to bind sec dsim bridge: -517
[    6.571316] lt8912 1-0048: failed to find dsi host