Freescale/meta-freescale #1804: build fails on meta-freescale/kirkstone due to missing revision in management-console upstream

0xharshal, 2024-04-25:
Suddenly the builds started failing.
Upon checking I notice that qoriq-mc-binary repository (https://github.com/nxp-qoriq/qoriq-mc-binary) no longer contains the revision mentioned in `meta-freescale/recipes-dpaa2/management-complex/management-complex_10.35.0.bb` https://github.com/Freescale/meta-freescale/blob/73a86058f0d44494dd0b17ee6dfbc4a5fb003fa8/recipes-dpaa2/management-complex/management-complex_10.35.0.bb#L10

The qoriq-mc-binary has some very recent changes and
The license file also doesn't match.

Can you shed light on this, is this intended or the repository is compromised?

Build log snippet showing `FetchError`
```
$ MACHINE="tqmlx2160a-mblx2160a" DISTRO="dumpling" bitbake management-complex                                                            
...
NOTE: Executing Tasks
WARNING: management-complex-10.35.0-r0 do_fetch: Failed to fetch URL git://github.com/nxp/qoriq-mc-binary;protocol=https;nobranch=1, attempting MIRRORS if available
ERROR: management-complex-10.35.0-r0 do_fetch: Fetcher failure: Unable to find revision 836d7bf606b51c18b5e3726d68dcf3b3ac4de74a in branch  even from upstream
ERROR: management-complex-10.35.0-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/nxp/qoriq-mc-binary;protocol=https;nobranch=1')
ERROR: Logfile of failure stored in: /Apr25_1435/build/tmp/work/tqmlx2160a_mblx2160a-tq-linux/management-complex/10.35.0-r0/temp/log.do_fetch.2844294
ERROR: Task (/Apr25_1435/meta-freescale/recipes-dpaa2/management-complex/management-complex_10.35.0.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 514 tasks of which 513 didn't need to be rerun and 1 failed.
```

alberhung, 2024-04-26:
I also faced the same issue a few days.
Any solution?

otavio, 2024-04-26:
Would you be willing to select and apply the necessary fixes from the master branch and prepare a pull request?

ryathnah, 2024-05-16:
Hello team,
let me know if any solution for this.