Freescale/meta-freescale #1823: imx93-evk: Failure to detect the SD card

fabioestevam, 2024-05-14:
Hi,

I built a core-image-minimal using the scarthgap from meta-freescale and flashed the resultant wic.gz into the SD card.

The kernel boots, but the SD card is not detected.

Does anyone have any ideas?

```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.28-lf-6.6.y-lf-6.6.y-g776652a165f5 (oe-user@oe-host) (aarch64-fslc-linux-gcc (GCC) 13.2.0, GNU ld (GNU Binutils) 2.42.0.20240216) #1 SMP PREEMPT Tue Apr 30 17:02:34 UTC 2024
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: NXP i.MX93 11X11 EVK board
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000b0000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000b0000000..0x00000000bfffffff (262144 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000002021e000..0x000000002021efff (4 KiB) nomap non-reusable rsc-table@2021e000
[    0.000000] OF: reserved mem: 0x0000000096000000..0x0000000097dfffff (30720 KiB) nomap non-reusable optee_core@96000000
[    0.000000] OF: reserved mem: 0x0000000097e00000..0x0000000097ffffff (2048 KiB) nomap non-reusable optee_shm@97e00000
[    0.000000] OF: reserved mem: 0x00000000a4000000..0x00000000a4007fff (32 KiB) nomap non-reusable vdev0vring0@a4000000
[    0.000000] OF: reserved mem: 0x00000000a4008000..0x00000000a400ffff (32 KiB) nomap non-reusable vdev0vring1@a4008000
[    0.000000] OF: reserved mem: 0x00000000a4010000..0x00000000a4017fff (32 KiB) nomap non-reusable vdev1vring0@a4010000
[    0.000000] OF: reserved mem: 0x00000000a4018000..0x00000000a401ffff (32 KiB) nomap non-reusable vdev1vring1@a4018000
[    0.000000] Reserved memory: created DMA memory pool at 0x00000000a4020000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node vdevbuffer@a4020000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000a4020000..0x00000000a411ffff (1024 KiB) nomap non-reusable vdevbuffer@a4020000
[    0.000000] Reserved memory: created DMA memory pool at 0x00000000a4120000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node ele-reserved@a4120000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000a4120000..0x00000000a421ffff (1024 KiB) nomap non-reusable ele-reserved@a4120000
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000c0000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node ethosu_region@C0000000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000c0000000..0x00000000cfffffff (262144 KiB) map reusable ethosu_region@C0000000
[    0.000000] earlycon: lpuart32 at MMIO32 0x0000000044380000 (options '')
[    0.000000] printk: bootconsole [lpuart32] enabled
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xffbc76c0-0xffbc9fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000095ffffff]
[    0.000000]   node   0: [mem 0x0000000098000000-0x00000000a3ffffff]
[    0.000000]   node   0: [mem 0x00000000a4000000-0x00000000a421ffff]
[    0.000000]   node   0: [mem 0x00000000a4220000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 8192 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 22 pages/cpu s50536 r8192 d31384 u90112
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: kernel page table isolation forced OFF by mitigations=off
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyLP0,115200 earlycon root=/dev/mmcblk1p2 rootwait rw
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 507904
[... 323 lines cut ...]
[    2.620183] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
[    2.624757] ALSA device list:
[    2.634820]   No soundcards found.
[    2.638628] Waiting for root device /dev/mmcblk1p2...
[   12.715863] platform regulator-dvdd_sel: deferred probe pending
[   12.721821] i2c 1-0022: deferred probe pending
[   12.726264] i2c 2-0050: deferred probe pending
[   12.730702] platform regulator-dvdd: deferred probe pending
[   12.736265] platform regulator-vdd: deferred probe pending
[   12.741750] platform regulator-vddo: deferred probe pending
[   12.747316] platform regulator-vaa_sel: deferred probe pending
[   12.753138] platform regulator-avdd: deferred probe pending
[   12.758703] i2c 1-0025: deferred probe pending
[   12.763139] i2c 2-0051: deferred probe pending
[   12.767581] i2c 2-003c: deferred probe pending
[   12.772014] platform imx93-lpm: deferred probe pending
[   12.777144] platform 42590000.serial: deferred probe pending
[   12.782795] platform usdhc3_pwrseq: deferred probe pending
[   12.788272] platform 42860000.mmc: deferred probe pending
[   12.793663] platform 428b0000.mmc: deferred probe pending
[   12.799054] platform sound-bt-sco: deferred probe pending
[   12.804444] platform regulator-usdhc2: deferred probe pending
[   12.810181] platform regulator-vdd-12v: deferred probe pending
[   12.816009] platform regulator-usdhc3: deferred probe pending
```

fabioestevam, 2024-05-14:
Also: if I manually build kernel lf-6.6.3-1.0.0 from NXP and flash it to the SD card, the kernel boots, the SD card is detected and the rootfs is successfully mounted.

It seems there is a kernel regression when going from 6.6.3 to 6.6.28.

MrCry0, 2024-05-15:
@fabioestevam Could you please add the boot log from NXP kernel?

fabioestevam, 2024-05-15:
Hi @MrCry0 

Here it goes:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.3-gccf0a99701a7 (fabio@fabio-Precision-3551) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #11 SMP PREEMPT Tue May 14 17:50:59 -03 2024
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: NXP i.MX93 11X11 EVK board
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000b0000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000b0000000..0x00000000bfffffff (262144 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000002021e000..0x000000002021efff (4 KiB) nomap non-reusable rsc-table@2021e000
[    0.000000] OF: reserved mem: 0x0000000096000000..0x0000000097dfffff (30720 KiB) nomap non-reusable optee_core@96000000
[    0.000000] OF: reserved mem: 0x0000000097e00000..0x0000000097ffffff (2048 KiB) nomap non-reusable optee_shm@97e00000
[    0.000000] OF: reserved mem: 0x00000000a4000000..0x00000000a4007fff (32 KiB) nomap non-reusable vdev0vring0@a4000000
[    0.000000] OF: reserved mem: 0x00000000a4008000..0x00000000a400ffff (32 KiB) nomap non-reusable vdev0vring1@a4008000
[    0.000000] OF: reserved mem: 0x00000000a4010000..0x00000000a4017fff (32 KiB) nomap non-reusable vdev1vring0@a4010000
[    0.000000] OF: reserved mem: 0x00000000a4018000..0x00000000a401ffff (32 KiB) nomap non-reusable vdev1vring1@a4018000
[    0.000000] Reserved memory: created DMA memory pool at 0x00000000a4020000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node vdevbuffer@a4020000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000a4020000..0x00000000a411ffff (1024 KiB) nomap non-reusable vdevbuffer@a4020000
[    0.000000] Reserved memory: created DMA memory pool at 0x00000000a4120000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node ele-reserved@a4120000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000a4120000..0x00000000a421ffff (1024 KiB) nomap non-reusable ele-reserved@a4120000
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000c0000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node ethosu_region@C0000000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000c0000000..0x00000000cfffffff (262144 KiB) map reusable ethosu_region@C0000000
[    0.000000] earlycon: lpuart32 at MMIO32 0x0000000044380000 (options '')
[    0.000000] printk: bootconsole [lpuart32] enabled
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xffbc66c0-0xffbc8fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000095ffffff]
[    0.000000]   node   0: [mem 0x0000000098000000-0x00000000a3ffffff]
[    0.000000]   node   0: [mem 0x00000000a4000000-0x00000000a421ffff]
[    0.000000]   node   0: [mem 0x00000000a4220000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 8192 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 22 pages/cpu s50536 r8192 d31384 u90112
[    0.000000] pcpu-alloc: s50536 r8192 d31384 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyLP0,115200 earlycon root=/dev/mmcblk1p2 rootwait rw
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 507904
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 2.
[    0.000000] software IO TLB: mapped [mem 0x00000000f9800000-0x00000000fd800000] (64MB)
[    0.000000] Memory: 1396520K/2064384K available (20672K kernel code, 1612K rwdata, 7628K rodata, 3904K init, 634K bss, 143576K reserved, 524288K cma-reserved)
[... 370 lines cut ...]
[    1.306123] mmc0: SDHCI controller on 42850000.mmc [42850000.mmc] using ADMA
[    1.377658] mmc0: new HS400 Enhanced strobe MMC card at address 0001
[    1.384515] mmcblk0: mmc0:0001 DA6016 14.7 GiB
[    1.390063]  mmcblk0: p1 p2
[... cut ...]
[    2.808579] mmc1: new ultra high speed DDR50 SDHC card at address aaaa
[    2.815581] mmcblk1: mmc1:aaaa SL16G 14.8 GiB
[    2.825196]  mmcblk1: p1 p2
[    2.851601] EXT4-fs (mmcblk1p2): recovery complete
[    2.857953] EXT4-fs (mmcblk1p2): mounted filesystem 436aa69d-b436-4ea1-af36-547c1218a920 r/w with ordered data mode. Quota mode: none.
[    3.564882] EXT4-fs (mmcblk1p2): re-mounted 436aa69d-b436-4ea1-af36-547c1218a920 r/w. Quota mode: none.
````

Does the SD card get detected for you when using 6.6.28?

Thanks

fabioestevam, 2024-06-04:
Hi @MrCry0 and @otavio 

I fixed the boot issue and submitted a pull request: 

https://github.com/Freescale/linux-fslc/pull/672/commits/e9ea6badf7e56126f0ce82a423c32e8b5715a4ea

fabioestevam, 2024-06-04:
Closing this issue as I sent a fix for this problem.

closed 2024-06-04 by fabioestevam

Livius90, 2024-09-07:
> Hi @MrCry0 and @otavio
> 
> I fixed the boot issue and submitted a pull request:
> 
> [Freescale/linux-fslc@e9ea6ba](https://github.com/Freescale/linux-fslc/commit/e9ea6badf7e56126f0ce82a423c32e8b5715a4ea)

I faced this same issue, but as i see it is just a missing GPIO driver problem? USDHC2 has a GPIO controlled vcc supply dependency in dts file of i.MX93 EVK board. It was my first suspect for this issue that something is not well with the probing of this GPIO.

Unfortunately this commit is not part of the latest linux-fslc v6.6.36 which is used in yocto recipe. Can you migrate it? By the way i can not se any defconfig for i.MX93 in the linux kernel repo, is the [imx_v8_defconfig](https://github.com/nxp-imx/linux-imx/blob/lf-6.6.y/arch/arm64/configs/imx_v8_defconfig) of i.MX8 is compatible to use for i.MX93 also?

Helyos96, 2024-11-08:
I am also facing that issue even with your fix included, using meta-freescale/scarthgap & linux-fslc-imx. Booting from sdcard is fine until mounting rootfs fails, as mmcblk1 is never found by the kernel.

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.54-yocto-evk-lf-6.6.y-gea326ecb9fc0 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 13.3.0, GNU ld (GNU Binutils) 2.42.0.20240723) #1 SMP PREEMPT Wed Nov  6 15:40:58 UTC 2024
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: NXP i.MX93 11X11 EVK board
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000b0000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000b0000000..0x00000000bfffffff (262144 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000002021e000..0x000000002021efff (4 KiB) nomap non-reusable rsc-table@2021e000
[    0.000000] OF: reserved mem: 0x0000000096000000..0x0000000097dfffff (30720 KiB) nomap non-reusable optee_core@96000000
[    0.000000] OF: reserved mem: 0x0000000097e00000..0x0000000097ffffff (2048 KiB) nomap non-reusable optee_shm@97e00000
[    0.000000] OF: reserved mem: 0x00000000a4000000..0x00000000a4007fff (32 KiB) nomap non-reusable vdev0vring0@a4000000
[    0.000000] OF: reserved mem: 0x00000000a4008000..0x00000000a400ffff (32 KiB) nomap non-reusable vdev0vring1@a4008000
[    0.000000] OF: reserved mem: 0x00000000a4010000..0x00000000a4017fff (32 KiB) nomap non-reusable vdev1vring0@a4010000
[    0.000000] OF: reserved mem: 0x00000000a4018000..0x00000000a401ffff (32 KiB) nomap non-reusable vdev1vring1@a4018000
[    0.000000] Reserved memory: created DMA memory pool at 0x00000000a4020000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node vdevbuffer@a4020000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000a4020000..0x00000000a411ffff (1024 KiB) nomap non-reusable vdevbuffer@a4020000
[    0.000000] Reserved memory: created DMA memory pool at 0x00000000a4120000, size 1 MiB
[    0.000000] OF: reserved mem: initialized node ele-reserved@a4120000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000a4120000..0x00000000a421ffff (1024 KiB) nomap non-reusable ele-reserved@a4120000
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000c0000000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node ethosu_region@C0000000, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000c0000000..0x00000000cfffffff (262144 KiB) map reusable ethosu_region@C0000000
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xffbc76c0-0xffbc9fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000095ffffff]
[    0.000000]   node   0: [mem 0x0000000098000000-0x00000000a3ffffff]
[    0.000000]   node   0: [mem 0x00000000a4000000-0x00000000a421ffff]
[    0.000000]   node   0: [mem 0x00000000a4220000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 8192 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.4
[    0.000000] percpu: Embedded 22 pages/cpu s50920 r8192 d31000 u90112
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: root=/dev/mmcblk1p1
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 507904
[    0.000000] Policy zone: DMA
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 2.
[    0.000000] software IO TLB: mapped [mem 0x00000000f9800000-0x00000000fd800000] (64MB)
[    0.000000] Memory: 1395756K/2064384K available (21120K kernel code, 1642K rwdata, 7804K rodata, 4032K init, 643K bss, 144340K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=2.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[... cut ...]
[    3.981516]  kernel_init_freeable+0x244/0x284
[    3.985864]  kernel_init+0x24/0x1dc
[    3.989350]  ret_from_fork+0x10/0x20
[    3.992923] SMP: stopping secondary CPUs
[    3.996844] Kernel Offset: disabled
[    4.000322] CPU features: 0x0,c0000000,40028143,1000721b
[    4.005626] Memory Limit: none
[    4.008682] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]---
```

Doesn't work either with distributed NXP kernel in LF_v6.6.36-2.1.0_images_IMX93EVK.zip. Will try and investigate..