Freescale/meta-freescale #1837: IW612 (WiFi) not working on i.MX93EVK

andreluizeng, 2024-06-05:
Hello !

I am trying to test the WiFi module (IW612) available on i.MX93EVK and it fails when loading the mlan module. 
The Issue is detected on both Nanbield and Scarthgap releases.

Steps to reproduce:

1 - Edit/create the wpa_supplicant.conf fiile with the following information:

          ctrl_interface=/var/run/wpa_supplicant
          ctrl_interface_group=0
          update_config=1
          
          network={
                 ssid="SSID"
                 scan_ssid=1
                 psk="password"
                 #proto=RSN
                 proto=WPA2
                 key_mgmt=WPA-PSK
                 pairwise=CCMP
                 group=CCMP
          }

2 - Probe the mlan and moal modules:

                # modprobe mlan
                # modprobe moal mod_para=nxp/wifi_mod_para.conf

3 - Configure the mlan0:
                
                # wpa_supplicant -d -B -i mlan0 -c wpa_supplicant.conf -Dnl80211 & 
                # udhcpc -i mlan0

Below the logs of a test performed on the master branch:

                root@imx93-11x11-lpddr4x-evk:~# uname -a                                        
                Linux imx93-11x11-lpddr4x-evk 6.6.28-lf-6.6.y-lf-6.6.y-gd8b0bac40433 #1 SMP PREx
                root@imx93-11x11-lpddr4x-evk:~# 

Output log:

                root@imx93-11x11-lpddr4x-evk:~# modprobe mlan                                   
                [   34.527752] mlan: loading out-of-tree module taints kernel.                  
                root@imx93-11x11-lpddr4x-evk:~# 

                root@imx93-11x11-lpddr4x-evk:~# modprobe moal mod_para=nxp/wifi_mod_para.conf   
                [   27.751392] wlan: Loading MWLAN driver                                       
                [   27.755533] wlan: Register to Bus Driver...                                  
                [   27.759933] vendor=0x0471 device=0x0205 class=0 function=1                   
                [   27.765480] Attach moal handle ops, card interface type: 0x109               
                [   27.771322] rps set to 0 from module param                                   
                [   27.776748] SDIW612: init module param from usr cfg                          
                [   27.781687] card_type: SDIW612, config block: 0                              
                [   27.786227] cfg80211_wext=0xf                                                
                [   27.789192] max_vir_bss=1                                                    
                [   27.791815] cal_data_cfg=none                                                
                [   27.794790] ps_mode = 1                                                      
                [   27.797237] auto_ds = 1                                                      
                [   27.799681] host_mlme=enable                                                 
                [   27.802559] fw_name=nxp/sduart_nw61x_v1.bin.se                               
                [   27.807028] SDIO: max_segs=128 max_seg_size=65535                            
                [   27.811726] rx_work=1 cpu_num=2                                              
                [   27.814858] Enable moal_recv_amsdu_packet                                    
                [   27.818891] Attach mlan adapter operations.card_type is 0x109.               
                [   27.825011] wlan: Enable TX SG mode                                          
                [   27.828529] wlan: Enable RX SG mode                                          
                [   27.835022] Request firmware: nxp/sduart_nw61x_v1.bin.se                     
                [   28.031573] Wlan: FW download over, firmwarelen=1004248 downloaded 916048    
                [   34.038943] Fail to poll firmware status: firmwarestat=0x0                   
                [   34.044444] FW failed to be active in time!                                  
                [   34.048627] wlan_dnld_fw fail ret=0xffffffff                                 
                [   34.052896] WLAN: Fail download FW with nowwait: 0                           
                [   34.057971] SDIO Func0 (0x0-0x9): 43 03 02 02 03 00 00 02 03 00              
                [   34.064097] SDIO Func1 (0x10-0x17): 00 00 00 00 00 00 00 00                  
                [   34.069947] SDIO Func1: (0x8) 00 (0x58) 00 (0x5c) 0d (0x5d) 00 (0x60) 01 (0x 
                [   34.083955] SDIO Func1 (0xe8-0xff): 00 00 00 00 50 fa 0d 00 24 14 70 c8 ab 1 
                [   34.194076] SDIO Func1 (0xe8-0xff): 00 00 00 00 50 fa 0d 00 24 14 70 c8 ab 1 
                [   34.203989] woal_request_fw failed                                           
                [   34.207388] Firmware Init Failed                                             
                [   34.211170] Free module params                                               
                [   34.214225] woal_add_card failed                                             
                [   34.217569] wlan_sdio: probe of mmc2:0001:1 failed with error -1             
                [   34.223738] wlan: Register to Bus Driver Done                                
                [   34.228120] wlan: Driver loaded successfully                                 
                root@imx93-11x11-lpddr4x-evk:~#    

                root@imx93-11x11-lpddr4x-evk:~# wpa_supplicant -d -B -i mlan0 -c wpa_supplicant.
                conf -Dnl80211 &                                                                
                root@imx93-11x11-lpddr4x-evk:~# wpa_supplicant v2.10                            
                random: Trying to read entropy from /dev/random                                 
                Successfully initialized wpa_supplicant                                         
                Initializing interface 'mlan0' conf 'wpa_supplicant.conf' driver 'nl80211' ctrl'
                Configuration file 'wpa_supplicant.conf' -> '/home/root/wpa_supplicant.conf'    
                Reading configuration file '/home/root/wpa_supplicant.conf'                     
                ctrl_interface='/var/run/wpa_supplicant'                                        
                ctrl_interface_group='0'                                                        
                update_config=1                                                                 
                Priority group 0                                                                
                   id=0 ssid='Andre-Router-1'                                                   
                Could not read interface mlan0 flags: No such device                            
                nl80211: Driver does not support authentication/association or connect commands 
                nl80211: deinit ifname=mlan0 disabled_11b_rates=0                               
                nl80211: Remove monitor interface: refcount=0                                   
                netlink: Operstate: ifindex=0 linkmode=0 (kernel-control), operstate=6 (IF_OPER)
                Could not read interface mlan0 flags: No such device                            
                mlan0: Failed to initialize driver interface                                    
                Failed to add interface mlan0                                                   
                mlan0: Cancelling scan request                                                  
                mlan0: Cancelling authentication timeout                                        
                Off-channel: Clear pending Action frame TX (pending_action_tx=(nil)             
                QM: Clear all active DSCP policies                                              
                mlan0: CTRL-EVENT-DSCP-POLICY clear_all                                         
                root@imx93-11x11-lpddr4x-evk:~# 

At this point the mlan0 is not available.

I appreciate any help.

regards,
Andre

Toomoch, 2024-06-12:
I have the issue in the `6.6.3-6.6.3-1.0.0-6.6.3-1.0.0-gc5bdfbdac605` Scarthgap version.

hcl-dr, 2024-09-30:
Caused by race with Bluetooth driver FW loading

hcl-dr, 2024-09-30:
Fix this using a config file in /etc/modprobe.d

> options moal mod_para=nxp/wifi_mod_para.conf drvdbg=0x6 cntry_txpwr=1
> softdep btnxpuart pre: moal

This forces the WiFi FW to load before Bluetooth

fabioestevam, 2024-09-30:
Hi @hcl-dr 

Thanks for your suggestion.

Care to submit a formal patch so that IW612 could work out of the box on imx93-evk?

Thanks

hcl-dr, 2024-09-30:
This would be a patch on meta-freescale and depends on both the chip in use as well as user preferences for WiFi.
Suggest using an append file like

```
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

SRC_URI:append = " \
    file://moal.conf \
    file://your_wifi_conf.conf \
"

do_install:append () {
	install -d ${D}${sysconfdir}/modprobe.d
	install -m 0644 ${WORKDIR}/moal.conf ${D}${sysconfdir}/modprobe.d/
	install -d ${D}${libdir}/firmware/nxp
	install -m 0644 ${WORKDIR}/your_wifi_conf.conf ${D}${libdir}/firmware/nxp/
}

FILES:${PN} += "${sysconfdir} ${libdir}/firmware/nxp"
```

hcl-dr, 2024-09-30:
@fabioestevam : You could split the modprobe file in two and just ship the "softdep" part in meta-freescale

fabioestevam, 2024-10-03:
@hcl-dr 

In the report from Andre,  I don't see any Bluetooth-related error logs. Do you think the original error is caused by the lack of "softdep"?

Were you able to get Wifi working on the imx93-evk?

hcl-dr, 2024-10-03:
I'm using a iMX93QSB which should be same/same. It is working fine with the softdep in place.

andreluizeng, 2024-11-14:
@hcl-dr thanks for your suggestion, I tried it but didn't work, maybe a did something wrong... for a quick test I just need to create a moal.conf file in /etc/modprobe.d  and when I enter the command: modprobe moal mod_para=nxp/wifi_mod_para.conf it should create the wlan0 device ? This is what I did and didn't work, what else am  I missing ?

the moal.conf file has the following content (as you suggested):
options moal mod_para=nxp/wifi_mod_para.conf drvdbg=0x6 cntry_txpwr=1
softdep btnxpuart pre: moal

cheers,
Andre

andreluizeng, 2024-11-15:
Hi @hcl-dr , 

I was able to get it working, I got a typo on my wpa_supplicant.conf file, fixed and it worked with your suggestion.

thank you.