Freescale/meta-freescale #1845: Some OP-TEE tests fail on i.MX93

tprrt, 2024-06-13:
Some OP-TEE regression tests stall with BSP 6.6.3-1.1.0, OP-TEE 4.0.0.imx, from the Scarthgap branch, commit-id 19ffbd1a6bf1:

```
* regression_1001 Core self tests
o regression_1001.1 Core self tests
  regression_1001.1 OK
o regression_1001.2 Core dt_driver self tests
  regression_1001.2 OK
  regression_1001 OK
 
* regression_1002 PTA parameters
  regression_1002 OK
 
* regression_1003 Core internal read/write mutex
    Number of parallel threads: 6 (2 writers and 4 readers)
    Max read concurrency: 1
    Max read waiters: 1
    Mean read concurrency: 1
    Mean read waiting: 1
  regression_1003 OK
 
* regression_1004 Test User Crypt TA

^CE/TC:? 0 get_rpc_alloc_res:645 RPC allocation failed. Non-secure world result: ret=0xffff000c ret_origin=0x2
D/TC:? 0 ldelf_syscall_open_bin:167 res=0xffff000c
E/LD:  init_elf:486 sys_open_ta_bin(cb3e5ba0-adf1-11e0-998b-0002a5d5c51b)
E/TC:? 0 ldelf_init_with_ldelf:152 ldelf failed with res: 0xffff000c
D/TC:? 0 tee_ta_open_session:694 init session failed 0xffff000c

```

tprrt, 2024-06-13:
And at boot, the following error is raised:
```
E/TC:? 0 get_rpc_alloc_res:645 RPC allocation failed. Non-secure world result: ret=0xffff000c ret_origin=0x2
D/TC:? 0 ldelf_syscall_open_bin:167 res=0xffff000c
E/LD:  init_elf:486 sys_open_ta_bin(560c5231-71bc-476d-8c2e-4ba107991e72)
E/TC:? 0 ldelf_init_with_ldelf:152 ldelf failed with res: 0xffff000c
D/TC:? 0 tee_ta_open_session:694 init session failed 0xffff000c
F/TC:? 0 plat_prng_add_jitter_entropy:72 0x3E

```

sahilnxp, 2024-06-17:
This kind of issue comes often when tee-supplicant is not running on kernel side.
@tprrt Can you please help me with link of images on which this issue is seen?

tprrt, 2024-06-19:
Hello @sahilnxp,

Yes, your distro doesn't use systemd as init manager and your recipe doesn't provide sysvinit support, like the meta-arm's optee-client recipe.

Kind regards,
Thomas

sahilnxp, 2024-06-19:
Can you start the tee-supplicant manually and then run xtest just to verify if this is tee-supplicant issue only ?
There is some issue we had some time back wrt tee-supplicant not starting automatically, Will discuss with @thochstein for a fix with this.

thochstein, 2024-06-19:
Hi @tprrt, we use systemd for our meta-imx distro. As I remember, systemd dropped sysvinit support on Scarthgap, and I added the following in order to start tee-supplicant.

https://github.com/nxp-imx/meta-imx/commit/a546b67f088b0207e3b19a5307ff6b889d488379

Would you happen to know how to adapt that for sysvinit?

tprrt, 2024-06-20:
Hello @thochstein ,

We don't use your meta-imx distro, especially images for manufacturing products.

Sysvinit is still supported as init manager in OE / Yocto.

Kind regards,
Thomas

closed 2024-06-20 by tprrt